luc-github/ESP3DLib #4: ESP3DLib Feedback

luc-github, 2020-01-15:
To link with https://github.com/luc-github/Marlin/issues/1
which will be locked soon

felixstorm, 2020-01-15:
@vivian-ng just fyi: I also had layer shifts a few days ago when I tried out a very recent Marlin build (https://github.com/MarlinFirmware/Marlin/commit/89ccd463d79007f5b1f15d4ffa8df913ae3a3373 from `bugfix-2.0.x`) and I had not changed anything on the hardware side. So there might really be some issue in the firmware or - another possibility - both our configs might just not be compatible with the current builds. I will probably try to look into it in the next few days, but as always will need to see when I find time for it.

vivian-ng, 2020-01-16:
[PR#16581](https://github.com/MarlinFirmware/Marlin/pull/16581) sent to add sanity check that throws an error when TMC SPI drivers are in use with MONITOR_DRIVER_STATUS and SDSUPPORT both enabled. The user will then need to decide which to disable (most probably MONITOR_DRIVER_STATUS since SDSUPPORT is essential for sending files for print via the webUI).

Other boards which also share SPI between SD card and TMC can simply define HAS_SHARED_SPI in their pins file to enable this check. I only updated the pins files for my boards since I don't think I am able to go through the schematics of all boards supported by Marlin to find out which are the boards with shared SPI.

@felixstorm Thank you for the input, hopefully you can find the issue. For me, the layer shifts are quite random. I can get a shift on a small print, but not a large one. It may have something to do with stepper timing or how jerk is implemented, but I can't say for sure without finding time to delve into the code.

luc-github, 2020-01-17:
the Development branch has now several fixes : 
```
Add websocket notification if any upload fail
Add more control during upload (SD/SPIFFS/Update)
Fix SD card > 4GB size calculation
Fix Apple Captive portal blocking websocket
Add auto mount for SD commands
Use latest embedded page
Display ESP3DLib Version in [ESP800] output
Force sleep mode to be none at start
```
To use development version just add `#devt` at the end of ESP3DLib git address in your platformio.ini

`ESP3DLib=https://github.com/luc-github/ESP3DLib.git#devt`   

today versions : UI: 2.1b56 / FW: bugfix-2.0.x / ESP3DLib: 1.0.3

felixstorm, 2020-01-17:
@vivian-ng It looks like I was able to avoid layer shifts by switching back to `CLASSIC_JERK` (https://github.com/felixstorm/Marlin/commit/fd50a136c539a2d9ec3c55e3682a3d940676aa9a). It looks like there are currently some issues in the interaction of Cura (I am still using 3.6), Ender 3 and Junction Deviation (new Marlin default), see e.g. https://github.com/MarlinFirmware/Marlin/issues/12403#issuecomment-570323589.

I only made one small test print so far, but the exact same G-code did have multiple layer shifts a few days ago with junction deviation enabled and now worked completely fine with classic jerk, so it's not a real proof but might be of help for you, too.

vivian-ng, 2020-01-17:
@felixstorm I have already gone back to classic jerk. But I will do a few more tests. Was using Cura 4.3 and it had been fine.
Also, for your Ender-3, how hot does the X and Y motors get? I am seeing them go up to around 45 degC when printing (with TMC2130 steppers set at 750mA).

@luc-github I have compiled with the devt branch, seems to be working. Will test more over the next few days.

felixstorm, 2020-01-17:
@vivian-ng Good to hear that layer shifts seem to be solved for you. I never really checked the temperature of the motors - I basically adjusted Vref on the TMC2208 drivers to match the currents somebody had published to be the correct ones for the Ender 3 it seems to be ok for the last few months. Also I do not really have a good way to measure the temperature on the motors, but I can just try to roughly estimate it using my fingers when I do the next longer print.

@vivian-ng I also compiled with the `devt` branch and it does seem to be working. Unfortunately I did not have enough time to test any details yet.  
One thing that happens in my setup is that refreshing the SD card directory from the web UI takes a while and then the ESP32 task watchdog gets triggered resulting in a reboot of Marlin. After the reboot, the correct SD card listing is then displayed in the web UI. Issuing `M20` through serial or as a web UI command works fine. If I can be of help with any further details on this please let me know.

I also created an issue / question in the Marlin repo asking where to best store WiFi credentials - if you have any opinion about this please feel free to comment: https://github.com/MarlinFirmware/Marlin/issues/16588

felixstorm, 2020-01-17:
@luc-github Sorry for the confusion: I just noticed that the reboot on refreshing the SD card only happens directly after power - any refresh thereafter seems to work fine. I am out of time now, but will do a few more tests on this and let you know the results...

luc-github, 2020-01-17:
@felixstorm I never got a restart after refreshing, it is true first listing is longer due to SDcard free space calculation.
if you got a stack decoded please share may need to add some WDT feed
Also one issue I have noticed in marlin, put one SD card of 2GB, do listing, then insert another SD of 16GB, do listing the capacity is still displayed as 2GB because of cache, this is something I must fix

luc-github, 2020-01-17:
About Wifi credential in configuration file it is ignored if you do not use them and also they are also ignored if they are stored in preferences NVS

luc-github, 2020-01-18:
I just received my new Creality LCD for the ESPE, I will work on adding support of notifications in lcd (https://github.com/luc-github/ESP3DLib/issues/5)  when back home next month

luc-github, 2020-01-23:
Actually I was able to use an esp32 + oled screen to set the lcd output - I did a PR to Marlin to add missing entry point to display properly IP at start on lcd https://github.com/MarlinFirmware/Marlin/pull/16658
as esplib is actually starting faster than Marlin in some case the IP was not displayed

WadeRobot, 2020-03-20:
marlin2.0.5  File path modification。
[image]

luc-github, 2020-03-20:
@WadeRobot fixed - thank you - seems I missed the monthly scheduled travis auto build report

jimhiggs, 2020-04-18:
Is this lib and the New Marlin ESP3D_WIFISUPPORT for ESP32 only?
I have an ESP01S (ESP8266) that was connected to my BIQU 3.5TFT display board and it works ok, but I wanted to put it on the main board a SKR Pro v1.1, mainly so i can upload files to the onboard sd by wifi. Both the SKR Pro and the TFT3.5 have sockets for an Esp01 but I don't know if something similar in pinout exists for ESP32

vivian-ng, 2020-04-18:
@jimhiggs 

> Is this lib and the New Marlin ESP3D_WIFISUPPORT for ESP32 only?
> I have an ESP01S (ESP8266) that was connected to my BIQU 3.5TFT display board and it works ok, but I wanted to put it on the main [...]

Yes, only if you are using ESP32 as the microcontroller for Marlin.

If you are using a WiFi module as a daughterboard, like the ESP01 that goes into a socket on your main board, please use [ESP3D](https://github.com/luc-github/ESP3D). The only thing in Marlin that needs to be changed in `Configuration.h` is to make sure
```
/**
 * Select a secondary serial port on the board to use for communication with the host.
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
//#define SERIAL_PORT_2 -1
```
that you have this set up properly for whatever main board you are using. Basically, uncomment the `#define` statement, and change `-1` to whatever number your board uses for the UART that is connected to the socket for the WiFi daughterboard.

jimhiggs, 2020-04-18:
Oh I see, thanks for the explanation. I wrongly assumed it meant an esp wifi add on. I did as you suggested and now have the esp3d working from the main board. Still doesn't help me as I still cant see the files on the SD. The pro has an onboard sd slot and an sd slot in the TFT and I was hoping ESP01 would at least be able to see one or other SD card, but nothing marlin crashed when I tried to upload a file from esp3d to the mainboard SD. Still I don't expect any of these issues are down to esp3d. Also ran into the 8.3 error nd I searched and found you comments about these being an issue with Marlin and not technically a bug.  It does rather make it useless though if you can't upload files with the gcode extension!

luc-github, 2020-04-18:
rename your file and it will work

jimhiggs, 2020-04-18:
If you mean make the filename 8.3 I did try renaming to copy.gco and using EPS3D to upload and it still failed. ESP3D reported lost connection and on the TFT it reported Upload failed.

jimhiggs, 2020-04-18:
I think there are other SKR Pro issues here I have to sort out first. Thanks for quick response.

luc-github, 2020-04-18:
pls feel free to open issue in esp3d repository if you have issue, I got for several pb with btt products and working on it but usually reset board during upload is a power supply stability issue, esp8266 is very sensitive

vivian-ng, 2020-05-05:
`espcom.cpp` will result in error if compiling without a LCD controller.

```
void Esp3DCom::echo(const char * data)
{
    SERIAL_ECHO_START();
    SERIAL_ECHOLNPAIR("", data);
    if (strlen(data)) {
        ui.set_status(data);
    } else {
        ui.reset_status();
    }
}
```

The call to `ui.set_status(data)` will give error as there is no ui. Recommend to add a `#ifdef` statement to check if LCD is being used before calling `ui` related methods.

luc-github, 2020-05-05:
that is weird I have checked this in the past - I will have a look

luc-github, 2020-05-05:
@vivian-ng can you tell me how do you reproduce issue ?
I have tested on my current ESPA and I do not have error and I do not have display set neither

I will check with latest Marlin as there may be some changes

vivian-ng, 2020-05-05:
@luc-github Sorry, please ignore. Somehow, I cannot reproduce the error anymore. Probably some glitch somewhere on my system.

luc-github, 2020-05-05:
@vivian-ng to be honest I have tested this before and was surprised it did not need a #define, I was also expecting I would need a check - but because it was ok I did not investigated - 
`Do not fix what is not broken` 
But Murphy law is always happy to popup...

Note ; I have updated to latest Marlin and still no issue, if you can reproduce let me know

closed 2020-12-11 by luc-github