luc-github/ESP3DLib #8: [Question] Using additional hardware serial for ESP32 in Marlin

vivian-ng, 2020-01-27:
This is not totally related to ESP3DLib.
It is actually a question more about the ESP32 HAL in Marlin. I wonder if it is possible to use the other hardware serial ports in the ESP32. Right now, I am using GPIO1 and GPIO3 (I think it is hardware serial0) for the CH340 USB-to-UART connection. The ESP32 still has Serial1 and Serial2 ununsed.

1. Does ESP3DLib use any of the hardware serial ports for websocketserial?
2. Is it possible to set up another hardware serial in the HAL, so that it can be used to connect to TMC2209 stepper drivers in UART mode?

Background:
I just realized that the TMC2209 allows up to 4 slave addresses, which means up to 4 TMC2209 stepper drivers can share a single UART as long as they have different slave addresses. This means if we can get a hardware serial up, we can use up to 4 TMC2209 stepper drivers on ESP32 boards.

Default Serial1 pins are GPIO9 and GPIO10, while Serial2 defaults to GPIO16 and GPIO17. At the same time, ESP32 actually allows users to set the pins they want to use for hardware serial (through some kind of pin matrix) so it would be even better if, for example, we can have something like:
```
#define HARDWARE_SERIAL1_RX 20
#define HARDWARE_SERIAL1_TX 21
```
to freely define the pins to use for hardware serial.

Thoughts?

luc-github, 2020-01-27:
Hi ESP3Lib does not use any pin for serial2ocket, serial2socket is a virtual serial as Marlin allows only 2 serial output, 

yes definitly any pin can be used for serial on esp32, but I think total hardware serial number is limited to 3(tbc)
also software serial is not recommended due to lack of reliability, see dev status: https://github.com/espressif/arduino-esp32/issues/3484#issuecomment-556988656

luc-github, 2020-01-27:
@vivian-ng 
are you sure about your GPIO number ?
there is not GPIO20 I think : https://randomnerdtutorials.com/esp32-pinout-reference-gpios/ https://github.com/espressif/arduino-esp32/blob/master/variants/esp32/pins_arduino.h

vivian-ng, 2020-01-27:
@luc-github I understand the issue with software serial, which is why my question was about hardware serial and how to define the pins for it.

As for GPIO number, that was just an example. I just gave two numbers that came into mind, but if we manage to find a way to define hardware serial pins, I will likely use pins 21 and 22 since those are the two pins that I have unused.

If anyone has any suggestions on how to set hardware serial pins, please share.

luc-github, 2020-01-27:
ho ok, FYI I do this in ESP3D already : https://github.com/luc-github/ESP3D/blob/3.0/esp3d/src/modules/serial/serial_service.cpp#L79

luc-github, 2020-01-27:
@vivian-ng 

here sample code you can try (1) is hardware serial1 :
```
HardwareSerial Myserial(1);
Myserial.begin(115200, SERIAL_8N1, HARDWARE_SERIAL1_RX , HARDWARE_SERIAL1_TX );
Myserial.setRxBufferSize(128);
```
or keeping same name to avoid to create another instance

```
Serial1.begin(115200, SERIAL_8N1, HARDWARE_SERIAL1_RX , HARDWARE_SERIAL1_TX );
Serial1.setRxBufferSize(128);
```

vivian-ng, 2020-01-28:
@luc-github Any suggestion on where this sample code should go?
I tried putting it in HAL.cpp, under HAL_init(), but it doesn't seem to work. I can compile, but I am not able to communicate with the TMC2209.

```
void HAL_init() {
  i2s_init();
  #define HARDWARE_SERIAL1_RX 21
  #define HARDWARE_SERIAL1_TX 22
  Serial1.begin(115200, SERIAL_8N1, HARDWARE_SERIAL1_RX , HARDWARE_SERIAL1_TX );
  Serial1.setRxBufferSize(128);
}
```

In Configuration_adv.h, I have
```
  #define  X_SLAVE_ADDRESS 0
  #define  Y_SLAVE_ADDRESS 1

  #define X_HARDWARE_SERIAL Serial1
  #define Y_HARDWARE_SERIAL Serial1
```
(MS1 and MS2 for X and Y axes have been set to reflect the slave address for each)

luc-github, 2020-01-28:
wrong close - I may be a config issue and init order if dual begin I see in code 
in tmc_serial_begin() 
```
#ifdef X_HARDWARE_SERIAL
 X_HARDWARE_SERIAL.begin(TMC_BAUD_RATE);
```
so should be Serial1.begin(TMC_BAUD_RATE, SERIAL_8N1, HARDWARE_SERIAL1_RX , HARDWARE_SERIAL1_TX );
```
but I also see it can be overloaded
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/module/stepper/trinamic.cpp#L113-L115

so may be just overload in pins_MRR_ESPE.h  should be ok no need a new begin
`#define TMC_BAUD_RATE 115200, SERIAL_8N1, 20, 21`

and just 
```
#define  X_SLAVE_ADDRESS 0
#define  Y_SLAVE_ADDRESS 1

#define X_HARDWARE_SERIAL Serial1
#define Y_HARDWARE_SERIAL Serial1
```

and of course define driver as TMC2209

I did not tested  but would be a clean way if it works

vivian-ng, 2020-01-28:
@luc-github You are a genius! This single line
```
#define TMC_BAUD_RATE 115200, SERIAL_8N1, 21, 22
```
allowed me to set up hardware serial on pins 21 (SDA) and 22 (SCL) as RX and TX.

And I found out what went wrong. Starting up hardware serial in `HAL_init()` probably works, but my biggest mistake was not reading the [manual](https://github.com/bigtreetech/BIGTREETECH-TMC2209-V1.2/tree/master/manual) of the TMC2209 stepper driver board that I was using. I had assumed that the pins labelled PDN (there are two of them) were both connected to PDN, so I connected one pin to RX, and the other to TX. Well, it seems only one of the pins is connected to PDN, and if I want to use the other "PDN" pin, I need to shift a SMD resistor on the board.

In the end, I spliced together another cable using only one of the PDN pins. And tada! I managed to get the MRR ESPA to talk to the two TMC2209s. One on X at address 0, the other on Y at address 3. I have yet to test with all four slave addresses (I only have two TMC2209s), though.

closed 2020-01-28 by vivian-ng

luc-github, 2020-01-28:
excellent, happy you succeed, 😁