luc-github/ESP3DLib #9: [BUG] Cannot find U8glib.h when compiling in PlatformIO/VSCode

vivian-ng, 2020-01-27:
I received errors:
```
.pio/libdeps/esp32/ESP3DLib/src/../../../../../Marlin/src/lcd/dogm/u8g_fontutf8.h:11:20: fatal error: U8glib.h: No such file or directory
```
It seems that using
```
#include MARLIN_PATH(lcd/ultralcd.h)
```
doesn't allow the eventual `u8g_fontutf8.h` to properly include `#include <U8glib.h>`.

I had to add
```
#include <U8glib.h>
```
to `espcom.cpp` in order to get it to compile.

Files further downstream being included using
```
#include "filename"
```
seems to work fine when using the `MARLIN_PATH()` macro, but files further downstream being included using
```
#include <filename>
```
seems to be causing issues.

luc-github, 2020-01-27:
I will check tonight, path syntax may change if file is  from Marlin or external library

luc-github, 2020-01-27:
I reproduced it using CR10 display and MRR_ESPE, I have only tested with MRR_ESPA and oled and without LCD.

Working on it now

luc-github, 2020-01-27:
@vivian-ng 
can you try to add 
```
#if HAS_GRAPHICAL_LCD
    #include <U8glib.h>
#endif
```
in espcom.cpp like this

Compilation is ok on my side - but I do not have the hardware in hand to check is fine

```
#include MARLIN_PATH(HAL/HAL_ESP32/HAL.h)
#if HAS_GRAPHICAL_LCD
    #include <U8glib.h>
#endif
#include MARLIN_PATH(lcd/ultralcd.h)
```

vivian-ng, 2020-01-27:
@luc-github The file with problem is U8glib.h.

I will give it a try when I can, but I don't expect any problems since I basically did the same thing.

luc-github, 2020-01-27:
ho because I understood reading you that `#include <U8glib.h>` solve issue and after  `#include <filename>` and so `#include <U8glib>` causing issue
 so I was confusing as it works on my side so I guessed it was another file

Ok so I can push this to devt branch then

felixstorm, 2020-01-27:
Had the same issue - apparently PlatformIO does not detect that ESP3DLib needs U8glib. If we add `#include <U8glib.h>` somewhere, PlatformIO will add this lib's path also to the include path for ESP3DLib and then the includes further down will work ok.

luc-github, 2020-01-27:
fix pushed to devt https://github.com/luc-github/ESP3DLib/tree/devt

closed 2020-02-20 by stale[bot]

vivian-ng, 2020-03-28:
@luc-github The fix [here](https://github.com/luc-github/ESP3DLib/issues/9#issuecomment-578652578) is in the `devt` branch, but not in the `master` branch. Will you be pushing it to the `master` branch soon? Otherwise, we may need to update the Marlin `platformio.ini` to use the `devt` branch by default, because I keep getting include failures as VSCode cannot find the required files.

luc-github, 2020-03-28:
Done - sorry for the delay - I was busy with other projects and forget to push it when I fixed the Marlin 2.0.5 compilation issue