luc-github/ESP3DLib #13: Enabling WIFI_CUSTOM_COMMANDS failes to compile

wildwing1970, 2020-04-29:
Hi there,

i have a WEMOS Mega, which has an ESP8266 on Board! I installed Marlin 2.0.5.3 on the Arduino and ESP3D on the ESP!

If i enable WIFI_CUSTOM_COMMANDS in the Config and try to compile i get the following error

/tmp/ccaFAvkG.ltrans0.ltrans.o: In function `GcodeSuite::process_parsed_command(bool)':
<artificial>:(.text+0xb13e): undefined reference to `wifi_custom_command(char*)'
collect2: error: ld returned 1 exit status
Multiple libraries were found for "LiquidCrystal.h"
 Used: /home/wildwing/Downloads/arduino-1.8.10/libraries/LiquidCrystal
exit status 1

What am i doing wrong ???

Greets Franz

vivian-ng, 2020-04-29:
@wildwing1970 The ESP3DLib is for ESP32, not ESP8266. You will need a 3D printer control board based on the ESP32 microcontroller.

If you are using a 3D printer control board using other microcontrollers, and want to add WiFi connectivity with a daughterboard (either ESP32 or ESP8266), you can use the ESP3D firmware.

wildwing1970, 2020-04-30:
Ok Thank you so far! Well i flashed ESP3D Firmware to the ES8266 Part of the WEMOS Mega! How do i enable ESP3D Support in Marling Firmware? Since the Mega has 3 Serial ports, to which Port do i have to connect the ESP8266 to communicate with the Marlin FW?

vivian-ng, 2020-04-30:
@wildwing1970 Please see the [wiki of ESP3D](https://github.com/luc-github/ESP3D/wiki) for information on how to use ESP3D. This repository is for the ESP3DLib which is the integration of ESP3D into Marlin for ESP32 microcontrollers.

Also, before proceeding further, you may want to make sure you are able to use your board to actually control a 3D printer. ESP3D only adds a web UI to allow communication with the 3D printer control board via UART. Your printer needs to be working in the first place. From the board you are using, the pinout is probably different from the Arduino Mega board, so the normal RAMPS board is probably not going to work for you. My suggestion is to flash Marlin on your board to the ATMEGA2560 microcontroller part first, make sure you can control the board via USB, then worry about how to add the wifi portion via ESP8266. The ESP8266 should be connected via UART to the ATMEGA2560 on your board, if you can figure out which UART/Serial port that is, you just need to enable that serial port in Marlin's `Configuration.h`. I have no experience with the board you are using so I cannot be of more help than this.

luc-github, 2020-05-01:
I have updated the readme to clarify the usage of esp3dlib and target

I think issue can be closed now

closed 2020-05-01 by luc-github