luc-github/ESP3DLib #24: Compile failed when enable ESP3D_WIFISUPPORT under PlatformIO

Surrizu, 2021-01-21:
Hello.

I´m able to compile any Marlin 2.0.x under PlatformIO but when I uncomment  #define ESP3D_WIFISUPPORT, the compile fails.

I just would like the ESPboard to work with my 3D printer (Arduino 2560 + RAMPS1.4).

Please, I have no idea whats going wrong.

HERE THE TERMINAL CODE DURING "Almost" COMPILATION:

Microsoft Windows [versão 6.1.7601]
Copyright (c) 2009 Microsoft Corporation. Todos os direitos reservados.

C:\Users\Henrique\Downloads\Marlin-2.0.x>platformio run -e mega2560
Processing mega2560 (platform: atmelavr; board: megaatmega2560; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/megaatmega2560.html
PLATFORM: Atmel AVR (3.1.0) > Arduino Mega or Mega 2560 ATmega2560 (Mega 2560)
HARDWARE: ATMEGA2560 16MHz, 8KB RAM, 248KB Flash
DEBUG: Current (avr-stub) On-board (avr-stub, simavr)
PACKAGES:
 - framework-arduino-avr 5.1.0
 - toolchain-atmelavr 1.50400.190710 (5.4.0)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Library Manager: Installing git+https://github.com/luc-github/ESP3DLib.git
FileNotFoundError: [WinError 2] O sistema não pode encontrar o arquivo especificado:
  File "C:\Users\Henrique\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 170:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\Henrique\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Henrique\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 287:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Henrique\.platformio\platforms\atmelavr\builder\main.py", line 162:
    target_elf = env.BuildProgram()
  File "C:\Users\Henrique\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Henrique\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 62:
    env.ProcessProjectDeps()
  File "C:\Users\Henrique\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Henrique\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 140:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\Henrique\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Henrique\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1062:
    project.install_dependencies()
  File "C:\Users\Henrique\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 898:
    lm.install(spec)
  File "c:\users\henrique\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 49:
    spec, silent=silent, skip_dependencies=skip_dependencies, force=force
  File "c:\users\henrique\.platformio\penv\lib\site-packages\platformio\package\manager\library.py", line 86:
    force=force,
  File "c:\users\henrique\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 97:
    pkg = self.install_from_url(spec.url, spec, silent=silent)
  File "c:\users\henrique\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 141:
    vcs = VCSClientFactory.new(tmp_dir, url)
  File "c:\users\henrique\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 55:
    src_dir, remote_url, tag, silent
  File "c:\users\henrique\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 137:
    self.configure()
  File "c:\users\henrique\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 146:
    result = proc.exec_command([cls.command, "--exec-path"])
  File "c:\users\henrique\.platformio\penv\lib\site-packages\platformio\proc.py", line 113:
    p = subprocess.Popen(*args, **kwargs)
  File "C:\Users\Henrique\.platformio\python3\lib\subprocess.py", line 800:
    restore_signals, start_new_session)
  File "C:\Users\Henrique\.platformio\python3\lib\subprocess.py", line 1207:
    startupinfo)
=========================================================== [FAILED] Took 2.99 seconds ==========

Environment    Status    Duration
-------------  --------  ------------
mega2560       FAILED    00:00:02.989
===================================================== 1 failed, 0 succeeded in 00:00:02.989 =====

C:\Users\Henrique\Downloads\Marlin-2.0.x>echo "done" >C:\Users\Henrique\AppData\Local\Temp\ipc

C:\Users\Henrique\Downloads\Marlin-2.0.x>

vivian-ng, 2021-01-21:
@Surrizu ESP3DLib is for use with ESP32-based 3D printer boards. If you are using RAMPS, you just need [ESP3D](https://github.com/luc-github/ESP3D). 

Leave
```
//#define ESP3D_WIFISUPPORT
```
commented out because this is for ESP32-based 3D printer boards like [MRR ESPA](https://github.com/maplerainresearch/MRR_ESPA/) and [MRR ESPE](https://github.com/maplerainresearch/MRR_ESPE/). If you are not using such boards, follow instructions for ESP3D instead and not this library.

luc-github, 2021-01-21:
yes as mentioned in readme 
[image]

Your board is not an ESP92 board it is a Mega/Ramps so not the right target

closed 2021-01-21 by luc-github

Surrizu, 2021-01-21:
Aha! So I think that's solved. I will compile the code with commented "//#define ESP3D_WIFISUPPORT" to test the communication.

luc-github, 2021-01-21:
have fun 😸