luc-github/ESP3DLib #27: [BUG]<Compile failed using platform IO>

agoy507, 2021-03-21:
**Describe the bug**

Compile failed using platform IO with framework-arduinoespressif32 3.10005.210308 (1.0.5)

**Screenshots**
[image]

**Marlin Firmware:**
- Marlin Branch/Version:[e.g bugfix-2.0.x] 
- Any change done in Marlin code/configuration  

**ESP3DLib :**
- ESP3DLib Branch/Version:[e.g 2.1b14]  
- Web UI Version:[e.g 2.1b14]  
- Wifi mode:[e.g. STA]  
- Any change done in ESP3DLib code/configuration  

**Board used (please complete the following information):**
 - BOARD: [e.g. esp32dev] 
 - Flash size: [e.g. 4M]
 - partitions: [e.g. default]

**Additional context**
Add any other context about the problem here.
Wiring, decoded stack, etc...

luc-github, 2021-03-21:
you must not change the platformio.ini currently for esp32 core - I guess you did it for : https://github.com/luc-github/ESP3DLib/discussions/26
but it introduce another issue that is fixed in esp32 core git already 

I wait for the fix to be implemented to see if I still need to change in ESP3Dlib

agoy507, 2021-03-21:
ok. thanks for advice.
but i have advice, if printing and wifi have connection in or out, happen skip on stepper motor.
and he said there is an update from espressif to dual core on the arduino framework to handle network on other core cpu.

luc-github, 2021-03-21:
as you see in issue fix is already in core : https://github.com/espressif/arduino-esp32/issues/4905
so soon you can use the latest esp32 core

closed 2021-03-21 by agoy507