luc-github/ESP3DLib #29: [BUG]<Failed Compile using Platform IO>

agoy507, 2021-04-14:
**Describe the bug**
error compile.
Compiling .pio\build\esp32\lib3f5\ESP3DLib\web_server.cpp.o
.pio/libdeps/esp32/ESP3DLib/src/sd_ESP32.cpp: In member function 'bool ESP_SD::openDir(String)':
.pio/libdeps/esp32/ESP3DLib/src/sd_ESP32.cpp:324:32: error: 'class CardReader' has no member named 'getSd2Card'
     if (!sd_volume.init(&(card.getSd2Card()))) {
                                ^
Compiling .pio\build\esp32\lib3f5\ESP3DLib\wificonfig.cpp.o
At global scope:
cc1plus.exe: warning: unrecognized command line option '-Wno-register'
*** [.pio\build\esp32\lib3f5\ESP3DLib\sd_ESP32.cpp.o] Error 1.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
[image]
[Terminal Message.txt](https://github.com/luc-github/ESP3DLib/files/6310390/Terminal.Message.txt)
my configuration file
[Marlin-bugfix-2.0.x.zip](https://github.com/luc-github/ESP3DLib/files/6310399/Marlin-bugfix-2.0.x.zip)
Version
[image]

**Marlin Firmware:**
- Marlin lastest Version at april 14 2021 
- Any change done in Marlin code/configuration  

**ESP3DLib :**
- ESP3DLib Branch/Version:[e.g 2.1b14]  
- Web UI Version:[e.g 2.1b14]  
- Wifi mode:[e.g. STA]  
- Any change done in ESP3DLib code/configuration  

**Board used (please complete the following information):**
 - BOARD: [e.g. ESP32] 
 - Flash size: [e.g. 4M]
 - partitions: [e.g. default]

**Additional context**
wiring board https://github.com/felixstorm/Creality_Ender_3_ESP32_Board

luc-github, 2021-04-14:
did you tested with 2.7.2 as mentioned in readme ?

agoy507, 2021-04-14:
yes. I Was enable this feature followed in readme. so i attach file configuration. please tested to, and CMIIW.

luc-github, 2021-04-14:
2.7.2 is ok
[image]

agoy507, 2021-04-14:
Can you try test with marlin bugfix lastest version ?

luc-github, 2021-04-14:
Marlin always break everything - and do not care backward compatibility - I will work on next release only - I have no bandwidth to test every change in Marlin and try to fix something that may be broken few commits later
Sorry
You are free to see what is wrong and push PR but in that case it may not work with 2.7.2

agoy507, 2021-04-14:
ok. i will wait next release with your lib compability. thanks for your contribute.

closed 2021-04-14 by agoy507

luc-github, 2021-04-14:
if homing in wrong direction it is just a Marlin setting - check the home direction setting

agoy507, 2021-04-14:
OK. thanks for advice. i will check again marlin config. 
i think on bugfix marlin was fix my problem. cause the day before yesterday i was report this issue.

luc-github, 2021-04-14:
@agoy507 motor direction and homing direction can be changed in Marlin FW that is for sure
if it is homing and moving which are up side down, it can also be changed also by just fliping the motor connector on board if it is possible but do it only when power off of you may damage driver

and for what it worst the change that generate your issue just happened few hours ago actually : https://github.com/MarlinFirmware/Marlin/commit/138340ee99a852bc116e2a436932bed074624fde
But I will only fix it when next release is out if I do now master will be incompatible with current 2.7.2 so need to wait as I do not want to maintain several branch and may be as mentioned code may change again before next release