luc-github/ESP3DLib #31: There is no connection between the firmware and the WEB

sergerz, 2021-06-21:
There is a board on ESP32. Compiled the firmware based on Marlin 2.0.9 and Esp3dlib. But there is no connection between the WEB interface and the firmware. A similar GR BL firmware works on this board. What should I do to link Marlin to the WEB? Compilation in PlatformIO successful ESP 32 4 Mb.

luc-github, 2021-06-21:
Did you do what read me explain ?

sergerz, 2021-06-21:
No connection between WEB and Marlin.
When controlling the printer from the WEB, the motors do not turn, it does not react to limits, and it does not read the SD card.
Compilation in PlatformIO is successful. There are no errors.
Marlin Firmware 2.0.9
ESP3DLib V1.0.1 No changes were made.
WEB UI Version: 2.1B70
Wifi mode: AP

Board ESP32Dev
Flash size: 4Mb
partitions: default

Motor control via I2S

luc-github, 2021-06-21:
Did you do what read me explain : how to configure Marlin?

sergerz, 2021-06-21:
Changes have been made to the configuration files

luc-github, 2021-06-21:
can you share your changes ?

sergerz, 2021-06-21:
**Configuration.h:**

#define SERIAL_PORT 0
#define BAUDRATE 250000
//#define SERIAL_PORT_2 -1
//#define BAUDRATE_2 250000   // Enable to override BAUDRATE
#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_MRR_ESPE 
#endif

**Configuration_adv.h:**
/**
 * WiFi Support (Espressif ESP32 WiFi)
 */
//#define WIFISUPPORT         // Marlin embedded WiFi managenent
#define ESP3D_WIFISUPPORT   // ESP3D Library WiFi management (https://github.com/luc-github/ESP3DLib)

#if EITHER(WIFISUPPORT, ESP3D_WIFISUPPORT)
  #define WEBSUPPORT          // Start a webserver (which may include auto-discovery)
  #define OTASUPPORT          // Support over-the-air firmware updates
  #define WIFI_CUSTOM_COMMAND // Accept feature config commands (e.g., WiFi ESP3D) from the host

  //AUTHENTICATION_FEATURE: protect pages by login password.
#define AUTHENTICATION_FEATURE

There is a change in the connection of ESPO32 pins  to the hardware:
/**
 * Marlin 3D Printer Firmware
 * Copyright (c) 2020 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (c) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <https://www.gnu.org/licenses/>.
 *
 */
#pragma once

/**
 * MRR ESPE pin assignments
 * MRR ESPE is a 3D printer control board based on the ESP32 microcontroller.
 * Supports 5 stepper drivers (using I2S stepper stream), heated bed,
 * single hotend, and LCD controller.
 */

#include "env_validate.h"

#if EXTRUDERS > 2 || E_STEPPERS > 2
  #error "MRR ESPE only supports two E Steppers. Comment out this line to continue."
#elif HOTENDS > 1
  #error "MRR ESPE only supports one hotend / E-stepper. Comment out this line to continue."
#endif

#define BOARD_INFO_NAME      "MRR ESPE"
#define BOARD_WEBSITE_URL    "github.com/maplerainresearch/MRR_ESPE"
#define DEFAULT_MACHINE_NAME BOARD_INFO_NAME

//
// Limit Switches
[... 94 lines cut ...]
    //#define LCD_PINS_D5                    150
    //#define LCD_PINS_D6                    152
    //#define LCD_PINS_D7                    153

  #else

    #error "Only CR10_STOCKDISPLAY and REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER are currently supported. Comment out this line to continue."

  #endif

  #define BTN_EN1                              0
  #define BTN_EN2                             12
  #define BTN_ENC                             14

#endif // HAS_MARLINUI_U8GLIB

// Hardware serial pins
// Add the following to Configuration.h or Configuration_adv.h to assign
// specific pins to hardware Serial1 and Serial2.
// Note: Serial2 can be defined using HARDWARE_SERIAL2_RX and HARDWARE_SERIAL2_TX but
// MRR ESPA does not have enough spare pins for such reassignment.
//#define HARDWARE_SERIAL1_RX                 21
//#define HARDWARE_SERIAL1_TX                 22
//#define HARDWARE_SERIAL2_RX                  2
//#define HARDWARE_SERIAL2_TX                  4

luc-github, 2021-06-21:
obviously you did not do the change explained in readme https://github.com/luc-github/ESP3DLib#how-to-enable-
so of course it cannot communicate

lols, 2021-06-23:
I also have a esp and in short you need to
-Enable serial port 2
-Enable webui
-Connect to the wifi ssid
-upload the index.html file
-set to client station connect to your router ssid
-set a static ip and now can access it everytime

Since you can already access the website
What you did not do is you have serial port 2 disabled
Enable by removing //
#define SERIAL_PORT_2 -1

sergerz, 2021-06-24:
I did everything described, but there was no connection between Marlin and the WEB UI. The WEB UI interface itself opens, Marlin does not work at the same time. There is no reaction of the actuators to actions with the interface.

sergerz, 2021-06-24:
The similar GRBL software on this board works well.

sergerz, 2021-06-24:
The device is a slightly modified https://github.com/bdring/Grbl_Esp32

luc-github, 2021-06-24:
hardware is not limitation - esp3dlib work even on single esp32 board
it is a pure Marlin configuration issue
can you share :
1 - output of M115 from webUI, a screen shot would be great
2 - your configuration.h and your configuration_adv.h

sergerz, 2021-06-24:
Configuration.h
/**
 * Marlin 3D Printer Firmware
 * Copyright (c) 2020 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (c) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <https://www.gnu.org/licenses/>.
 *
 */
#pragma once

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 */
#define CONFIGURATION_H_VERSION 02000900

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
 * Here are some useful links to help get your machine configured and calibrated:
 *
 * Example Configs:     https://github.com/MarlinFirmware/Configurations/branches/all
 *
 * Průša Calculator:    https://blog.prusaprinters.org/calculator_3416/
 *
 * Calibration Guides:  https://reprap.org/wiki/Calibration
 *                      https://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
 *                      https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
 *                      https://youtu.be/wAL9d7FgInk
 *
 * Calibration Objects: https://www.thingiverse.com/thing:5573
 *                      https://www.thingiverse.com/thing:1278865
 */

//===========================================================================
//========================== DELTA / SCARA / TPARA ==========================
//===========================================================================
//
// Download configurations from the link above and customize for your machine.
// Examples are located in config/examples/delta, .../SCARA, and .../TPARA.
//
//===========================================================================

// @section info
[... 2780 lines cut ...]
 *  - Turn off after the print has finished and the user has pushed a button
 */
#if ANY(BLINKM, RGB_LED, RGBW_LED, PCA9632, PCA9533, NEOPIXEL_LED)
  #define PRINTER_EVENT_LEDS
#endif

/**
 * Number of servos
 *
 * For some servo-related options NUM_SERVOS will be set automatically.
 * Set this manually if there are extra servos needing manual control.
 * Set to 0 to turn off servo support.
 */
//#define NUM_SERVOS 3 // Servo index starts with 0 for M280 command

// (ms) Delay  before the next move will start, to give the servo time to reach its target angle.
// 300ms is a good value but you can try less delay.
// If the servo can't reach the requested position, increase it.
#define SERVO_DELAY { 300 }

// Only power servos during movement, otherwise leave off to prevent jitter
//#define DEACTIVATE_SERVOS_AFTER_MOVE

// Edit servo angles with M281 and save to EEPROM with M500
//#define EDITABLE_SERVO_ANGLES

sergerz, 2021-06-24:
Configurations_adv.h
/**
 * Marlin 3D Printer Firmware
 * Copyright (c) 2020 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (c) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <https://www.gnu.org/licenses/>.
 *
 */
#pragma once

/**
 * Configuration_adv.h
 *
 * Advanced settings.
 * Only change these if you know exactly what you're doing.
 * Some of these settings can damage your printer if improperly set!
 *
 * Basic settings can be found in Configuration.h
 */
#define CONFIGURATION_ADV_H_VERSION 02000900

//===========================================================================
//============================= Thermal Settings ============================
//===========================================================================
// @section temperature

/**
 * Thermocouple sensors are quite sensitive to noise.  Any noise induced in
 * the sensor wires, such as by stepper motor wires run in parallel to them,
 * may result in the thermocouple sensor reporting spurious errors.  This
 * value is the number of errors which can occur in a row before the error
 * is reported.  This allows us to ignore intermittent error conditions while
 * still detecting an actual failure, which should result in a continuous
 * stream of errors from the sensor.
 *
 * Set this value to 0 to fail on the first error to occur.
 */
#define THERMOCOUPLE_MAX_ERRORS 15

//
// Custom Thermistor 1000 parameters
//
#if TEMP_SENSOR_0 == 1000
  #define HOTEND0_PULLUP_RESISTOR_OHMS 4700    // Pullup resistor
  #define HOTEND0_RESISTANCE_25C_OHMS  100000  // Resistance at 25C
  #define HOTEND0_BETA                 3950    // Beta value
#endif

#if TEMP_SENSOR_1 == 1000
  #define HOTEND1_PULLUP_RESISTOR_OHMS 4700    // Pullup resistor
  #define HOTEND1_RESISTANCE_25C_OHMS  100000  // Resistance at 25C
  #define HOTEND1_BETA                 3950    // Beta value
#endif

#if TEMP_SENSOR_2 == 1000
  #define HOTEND2_PULLUP_RESISTOR_OHMS 4700    // Pullup resistor
[... 4068 lines cut ...]
//
// M42 - Set pin states
//
//#define DIRECT_PIN_CONTROL

//
// M43 - display pin status, toggle pins, watch pins, watch endstops & toggle LED, test servo probe
//
//#define PINS_DEBUGGING

// Enable Marlin dev mode which adds some special commands
//#define MARLIN_DEV_MODE

/**
 * Postmortem Debugging captures misbehavior and outputs the CPU status and backtrace to serial.
 * When running in the debugger it will break for debugging. This is useful to help understand
 * a crash from a remote location. Requires ~400 bytes of SRAM and 5Kb of flash.
 */
//#define POSTMORTEM_DEBUGGING

/**
 * Software Reset options
 */
//#define SOFT_RESET_VIA_SERIAL         // 'KILL' and '^X' commands will soft-reset the controller
//#define SOFT_RESET_ON_KILL            // Use a digital button to soft-reset the controller after KILL

sergerz, 2021-06-24:
I'm sorry that I'm not very skilled.

luc-github, 2021-06-24:
Ok reading now - next time please drop a zip file it is easier to get

luc-github, 2021-06-24:
Are you using MRR_ESPE ? you mentionned before you use Bart board or I misunderstood ?
can you share the boot serial output of Marlin, you just need to to select Monitor in VSCode
I am pretty sure you issue is : ` Error:Printer halted. kill() called!`

sergerz, 2021-06-24:
Ok

sergerz, 2021-06-24:
As a template, I use the configuration of the MRR ESPE board. This board is most suitable for the electrical circuit. Stepper motor drivers are connected to registers that allocate the I2S sequence. The board is made on the basis of the Bart board. If the printer is stopped, how can I start it? What should I do with the VSCode serial monitor

sergerz, 2021-06-24:
The card is not currently connected to the printer. There are 4 stepper motors connected to it, power supply.

luc-github, 2021-06-24:
As I mentioned before your issue is Marlin configuration, I do not know what you use but my guess the board definition you are using do not match your board, so thermistor is not detected and Marlin go to error because it cannot detect the temperature - I suggest you to configure first properly Marlin for your board  first, Marlin will stop everything if any error is detected

sergerz, 2021-06-24:
Thanks. I understood. I have a printer, I will connect it and then I will write about the results. But it will be in a couple of days.

luc-github, 2021-07-04:
@sergerz  any update ?

sergerz, 2021-07-06:
My existing board cannot fully work with a 3d printer due to the lack of a set of input and output modules. Now I'm preparing them for the order. Therefore, the testing is delayed.

closed 2021-08-03 by stale[bot]