luc-github/ESP3DLib #32: [FEATURE REQUEST]<Add Webdav>

agoy507, 2021-07-03:
Please add webdav feature. that feature will be very useful to save Gcode direct to sdcard.

luc-github, 2021-07-03:
it is already done in ESP3D and in roadmap for ESP3DLib https://github.com/luc-github/ESP3D/blob/3.0/ESP3D-features.xls

agoy507, 2021-07-07:
Hei luc.
I try to added webdav feature used your webdav library. file on LITTLEFS can showing on window explorer.
I tried to change LITTLEFS to SD_MMC like https://github.com/luc-github/ESP3D/issues/491#issuecomment-766157740 but it can't work. what's missing ?? can you explain ?

luc-github, 2021-07-07:
in esp3dlib I use Marlin sd lib not esp32 core sd lib

agoy507, 2021-07-07:
you mean Marlin sd lib is https://github.com/luc-github/ESP3DLib/blob/49eb92ba4fca575591e2042b2b25944d5770291b/src/sd_ESP32.h ?

luc-github, 2021-07-07:
yes, check the cpp file, it is  just  a wrapper

closed 2021-08-03 by stale[bot]