luc-github/ESP3DLib #39: [FEATURE REQUEST]ESP3DLib 2.0

luc-github, 2022-02-01:
Here features that will be covered by ESP3DLib 2.0 : 
- Webui 2.1 support (not 3.0 as not yet ready)
- Move to SSDP asyncudp library
- Move ESP32 core to 2.0.2 / 2.0.3
- Move from SPIFFS to LittleFS
- AP mode setup reviewed
- Add Telnet server support
- Add MKS Cura pluggin support for file upload
- Move SD support from Marlin SD Lib to SDFat 2.0 for better performance (TBC)

Testing boards:
1 - MRR_ESPA V1.3 without display
2 - FYSETC V1.0 + MKS TFT32_L V4.0 (With BTT FW)
3 - MKS TinyBee V1.0 + MKS Mini12864 V3.0
4 - Panda Zhu V1.2

Marlin Code base should be 2.0.9.3 bugfix  2.0.9.4? (TBD)

luc-github, 2022-02-15:
Some update:
1 - Use ESP3D 3.0 as code base is almost done - what is not yet implemented is the SD upload 
2 -  Use of  SSDP asyncudp library is implemented but I got some random (refering to esp_timer) crash and also some Watchdog reset on ESP3DLib task , even without the SSDP library so I need to dig in 
using `disableCore0WDT();` in Esp3DLib seems mitigating the issue for the moment but need to need to investigate
3 - Moving Marlin to latest ESP32 core git version is done (2.0.9.3+) - I did some change to make it support it in Marlin sources
4 - Moving from SPIFFS to LittleFS is also implemented 
5 - Telnet is implemented 
6 - Notiffications are implemented as part of ESP3D 3.0 code but not yet tested

I have currently only tested with MRR_ESPA V1.3 without display -  I still need to adjust code with display and with serial display

WebUI is finally the 3.0 as even it not yet finished it will be better to focus on latest versions 
since use of latest Marlin git seems supporting long name I will also spend time to add it 
```
#define LONG_FILENAME_HOST_SUPPORT    // Get the long filename of a file/folder with 'M33 <dosname>' and list long filenames with 'M20 L'
#define LONG_FILENAME_WRITE_SUPPORT   // Create / delete files with long filenames via M28, M30, and Binary Transfer Protocol
```
That said, I need to see how to handle Marlin with Long name support and Marlin without long name support in WebUI 3.0 refactoring

So it should save time to stay on WebUI 3.0 and ESP3DLib may directly jump to 3.0 as final version

Need to spend some time on stability issues before move to SD support then WebUI Update

luc-github, 2022-02-15:
Here the base for the display message output
[image]

Some screenshots

[image]

[image]
[image]
[image]
[image]

luc-github, 2022-03-02:
Ok final workaround is now validated instead of using task wdt feeder I use global wdt feeder and it seems ok after one week of long run test
Display output is now properly managed 
Now working on SD Support - I wish I can use the SDFat instead of Marlin SDLib which is very slow for upload - it seems Marlin support now long name in command but no more subdirectory files for print command: https://github.com/luc-github/ESP3D-WEBUI/issues/221#issuecomment-1055569968

luc-github, 2022-03-02:
Seems use current ESP3D V3 code using SDFat V2 fit with existing Marlin SDLib
```
>[ESP940]
Directory on SD : /
<DIR> 	overlays 	
<DIR> 	System Volume Information 	
<DIR> 	Archive 	
<DIR> 	New folder 	
      	bcm2708-rpi-b-plus.dtb 	13.94 KB 	
      	COPYING.linux 	18.25 KB 	
      	LICENCE.broadcom 	1.46 KB 	
      	bcm2709-rpi-2-b.dtb 	15.00 KB 	
      	bcm2708-rpi-b.dtb 	13.68 KB 	
      	bcm2708-rpi-cm.dtb 	13.64 KB 	
      	issue.txt 	145 B 	
      	CUBE.GCO 	205.31 KB 	
      	CUBE01.GCO 	1.99 MB 	
      	SUPPORT2.GCO 	4.40 MB 	
      	bootcode.bin 	17.51 KB 	
      	cmdline.txt 	136 B 	
      	config.txt 	1.62 KB 	
      	fixup.dat 	6.47 KB 	
      	fixup_cd.dat 	2.48 KB 	
      	fixup_db.dat 	9.52 KB 	
      	kernel.img 	3.94 MB 	
      	kernel7.img 	4.04 MB 	
      	fixup_x.dat 	9.52 KB 	
      	start_cd.elf 	619.66 KB 	
      	start_x.elf 	3.73 MB 	
      	LICENSE.oracle 	18.53 KB 	
      	bcm2710-rpi-3-b.dtb 	15.62 KB 	
      	bcm2710-rpi-cm3.dtb 	14.99 KB 	
24 file(s), 4 dir(s)
Total 62.95 MB, Used 24.63 MB, Available: 38.31 MB
ok
>M20
Begin file list
CUBE.GCO 210240
CUBE01.GCO 2089832
SUPPORT2.GCO 4613256
ARCHIVE/CUBE-C~1.GCO 210240
NEWFOL~1/SUPPORT2.GCO 4613256
End file list
ok
```

So now need to see how to avoid conflict in usage and allow only to access SD at once - but so far so good

luc-github, 2022-03-03:
Upgrade to SDFat 2.1.2 is now done and tested UTF file name also:
when it is good in ESP3DLib - even with latest bugfix and enabling
```
 #define UTF_FILENAME_SUPPORT
```
the output seems incorrect

```
Directory on SD : /
<DIR> 	overlays 	
<DIR> 	System Volume Information 	
<DIR> 	Archive 	
<DIR> 	New folder 	
      	bcm2708-rpi-b-plus.dtb 	13.94 KB 	
      	COPYING.linux 	18.25 KB 	
      	LICENCE.broadcom 	1.46 KB 	
      	bcm2709-rpi-2-b.dtb 	15.00 KB 	
      	bcm2708-rpi-b.dtb 	13.68 KB 	
      	bcm2708-rpi-cm.dtb 	13.64 KB 	
      	issue.txt 	145 B 	
      	CUBE.GCO 	205.31 KB 	
      	CUBE01.GCO 	1.99 MB 	
      	SUPPORT2.GCO 	4.40 MB 	
      	LEBOSSE.pdf 	0 B 	
      	bootcode.bin 	17.51 KB 	
      	cmdline.txt 	136 B 	
      	config.txt 	1.62 KB 	
      	fixup.dat 	6.47 KB 	
      	fixup_cd.dat 	2.48 KB 	
      	fixup_db.dat 	9.52 KB 	
      	kernel.img 	3.94 MB 	
      	kernel7.img 	4.04 MB 	
      	fixup_x.dat 	9.52 KB 	
      	VGMIL-6001.jpg 	1.69 MB 	
      	start_cd.elf 	619.66 KB 	
      	start_x.elf 	3.73 MB 	
      	LICENSE.oracle 	18.53 KB 	
      	bcm2710-rpi-3-b.dtb 	15.62 KB 	
      	Transfert.pdf 	57.88 KB 	
      	bcm2710-rpi-cm3.dtb 	14.99 KB 	
      	610203146-ddt.pdf 	7.30 MB 	
      	盧克test.GCO 	4.40 MB 	
      	6102031462-ddt.pdf 	7.30 MB 	
30 file(s), 4 dir(s)
Total 62.95 MB, Used 49.46 MB, Available: 13.48 MB
ok
Begin file list
CUBE.GCO 210240 CUBE.GCO
CUBE01.GCO 2089832 CUBE01.GCO
SUPPORT2.GCO 4613256 SUPPORT2.GCO
ARCHIVE/CUBE-C~1.GCO 210240 Archive/CUBE - Copy.G
NEWFOL~1/SUPPORT2.GCO 4613256 New folder/SUPPORT2.GCO
TEST~1.GCO 4613256 __test.GCO
End file list
ok
```
still need to test print command but seems not able to print sub folder file... TBC

WebUI is working as expected :
[image]

still need to handle possible sd access conflict in ESP3DLib
I am planning to do a SD release when ESP3DLib access SD and SD Mount when operation is done -> it will mimic the share SD behaviour => To be tested

luc-github, 2022-03-20:
The sharing SD code have been updated as well as the new format of ESPXXX output as explained here : https://github.com/luc-github/ESP3D-WEBUI/issues/157#issuecomment-1073143762
Still have FTP/WebDav SD access is not updated - I want to do the webui update first with updated output support and see how to handle the UTF / long name support in webUI as it seems it is still incomplete in Marlin : https://github.com/luc-github/ESP3D-WEBUI/issues/221#issuecomment-1058112630

luc-github, 2022-04-17:
Ok here we are - Webui V3 is now usable for Marlin-embedded but will still be updated : https://github.com/luc-github/ESP3D-WEBUI/blob/3.0-rewrite/dist/Printer3D/Marlin-embedded/index.html.gz?raw=true

to use ESP3DLib V2 with latest Marlin bug-fix and latest ESP32 core   you need currently to use this fork / branch : https://github.com/luc-github/Marlin/tree/ESP3DLibV2

here the board and status that I will updated during my tests
Board | Status  | notes
---|---|---
MRR_ESPA | Ok | No display
MRR_ESPE | Should be Ok | No board on hand but compilation Ok 
FYSECT_E4 | Ok | with BTT 2.8 serial TFT
PANDA Zhu V1.2 |   | Not tested yet
TinyBee V1 |    | Not tested yet 

I have already starting to test and found some cases that I could test on simulator 😉 so will update WebUI / ESP3DLib in parallele

I do not have time to write doc for the configuration / feature list at this moment if you want to help on it ping me so you can do it on Wiki

luc-github, 2022-05-04:
Issue with TinyBee MKS screen is now solved - issue was related to ESP3DLib code - so far so good 
the webUI is also under test with ESP3D V3 https://github.com/luc-github/ESP3D/issues/741
So I prefer to continue tests as ESP3DLib V2 and ESP3D V3 share same code

luc-github, 2022-05-09:
Finally WebDav is working as expected ^_^
[image]

I still have to review FTP server code in same way and should be ok

luc-github, 2022-05-10:
FTP server code is now updated - the number of connections must be limited to 1 as it does not support multiple request
[image]

agoy507, 2022-11-03:
Hi Luc. I have tried ESP3dLib v3.0 on my custom Board. But file gcode and the folder in which there is a gcode file not showing.
and controller fan for cooling main board  when stepper or heater on can't function.
when i used esp3dlib v1 with marlin 2.0.9 not happen problem.
 
[image]
[image]
[image]
[image]
[image]
[image]
[image]

luc-github, 2022-11-04:
@agoy507 please this is status thread, not issue ticket do not hijack it, I have moved it to another thicket

agoy507, 2022-11-04:
> @agoy507 please this is status thread, not issue ticket do not hijack it, I have moved it to another ticket

I'm sorry luc. I don't know. But i'm newbie on github

luc-github, 2023-12-23:
Because there was no realease for a while from Marlin, here the branch of latest bugfix-2.1.x with several fixes:  https://github.com/luc-github/Marlin/tree/ESP3DLib-V3-bugfix-2.1.x that support latest ESP3DLib 3.0, FYI upstream is broken for ESP3DLib 1.0 https://github.com/luc-github/ESP3DLib/issues/75

Current change log for reference:

* Update PIO to 6.1.0 in esp32.ini ( esp32-arduino core v2.0.7)
*  features.ini
    - Add ESP3D_WIFISUPPORT better discriminate needed libraries
    - Update NEOPIXEL_LED library to 1.12.0 to support new esp32-arduino core

* Update plateformio.ini
    - Update upload baud rate to 115200 as default (suggested per vivian-ng for MRR_ESPA  and MRR_ESPE)
    - Update monitor speed to 115200 as default
    - Add flash si8ze for each esp32 flasy that is not 4MB

* Update ESP32 HAL 
    - fastio to handle extended pins 
    - Update ADC channels naming to match new esp32-arduino core
    - Update config.auto_reload to match new esp32-arduino core
    - Fix TMC2209 error with new esp32-arduino core
    - Add SERIAL_PORT_3 feature support to ESP32
    - Add a new define `CUSTOM_SD_ACCESS` to allow concurrent SD access by external library
    - Update patch of u8g_esp32_spi.cpp to be more generic and no linked to ESP3D_WIFISUPPORT but to CUSTOM_SD_ACCESS

* Fix Configuration_adv.h to restore broken support of ESP3D_WIFISUPPORT

* Update cardreader.cpp to support a new define `CUSTOM_SD_ACCESS` to allow concurrent SD access by external library

* Add `#define OLD_ADAPTIVE_MULTISTEPPING` to pins_MKS_TINYBEE.h to fix deadloop at start

* Fix SanitiCheck.h to restore broken support of ESP3D_WIFISUPPORT

* Update pins_FYSETC_E4.h and MarlinCore.h to support `#define SKIP_TMC_SERIAL_BEGIN` to fix Serial initialization error if using TMC2209 in uart mode because of multiple Serial initialization that is already done in HAL

* Update .gitignore to ignore serial logs files

luc-github, 2024-01-26:
In same spirit of [ESP3D-TFT](https://esp3d.io/esp3d-tft/v1.x/hardware/index.html) I have listed all tested boards : https://esp3d.io/esp3dlib/v3.x/hardware/index.html
I did same for 1.0 https://esp3d.io/esp3dlib/v1.x/hardware/index.html

Feel free to let me know if need more data or I missed some boards

fermino, 2024-09-29:
Hey @luc-github, first of all thanks for all the work in this project, it's looking great!
I'd like to help a bit with this, is there any blocker that would prevent having the changes mentioned a couple of comments ago in Marlin upstream? If not, is it okay with you if I start sending some of the changes there as PRs? I'm thinking of making small, isolated PRs so there's a better chance to have them merged. 

Kind regards :)
Fermín.

luc-github, 2024-09-29:
I have to sync with latest ESP3D Code 
Why so hurry to push change upstream ?

luc-github, 2024-10-10:
@fermino did you already tested my fork with the changes ?

luc-github, 2024-12-09:
It has been a long time I did not sync Marlin and ESP3DLIb - as ESP3D will be soon move to beta state it is time to do I think - I will work on it this week

fermino, 2024-12-12:
Hey there! Sorry I totally missed the previous comments!

I haven't tested te changes yet, and right now I don't have the time to jump straight to it (it's also been a couple of months since I touched Marlin), but if there are some specific things that I can lend a hand with, I'll be happy to put in some time and help! :)

luc-github, 2024-12-21:
ESP3D and ESP3D-WEBUI are now in beta state, webui is now in translation correction state  from community - it is time now to do a new sync of :
- the Marlin fork to get latest update and use pioarduino instead of platformio to use idf 5.1.4 / core 3.0.4
- ESP3DLib code with ESP3D to be inline with ESP3D beta 1

I will start next week if no issue pop up