luc-github/ESP3DLib #43: [BUG]<ESP3DLib included in Marlin 2.0.9.3 - No media error>

ul42, 2022-05-05:


ul42, 2022-05-06:
Ok je note ça, se sera utile.

Voici ce que me renvois Pronterface :
Connecting...
Printer is now online.
echo:busy: processing
echo:busy: processing
echo:busy: processing
>>>M20
SENDING:M20
echo:No media
>>>M21
SENDING:M21
echo:No SD card
>>>M20
SENDING:M20
echo:No media

Je vais changer quelques paramètres et je vous recontacte.

luc-github, 2022-05-06:
Ben deja :
`echo:busy: processing`
si aucune commande n'est envoye ce n'est pas normal je pense

le No media  / No SD Card montre bien que Marlin ne voit pas la carte SD 
Quel est le pinout utilise pour la carte SD ?

ul42, 2022-05-06:
J'ai activé le paramètre :
#define SD_SPI_SPEED SPI_EIGHTH_SPEED

Et ça me donne ça :
[image]

J'ai également vérifié les brochages physique (ou cas ou) et tout est correctement relié a l'ESP32S.
Voici les brochages logiciels et un peu de doc en plus :
#define SD_MOSI_PIN                           23
#define SD_MISO_PIN                           19
#define SD_SCK_PIN                            18
#define SDSS                                   5
[image]
[image]

luc-github, 2022-05-06:
Je vous conseille de tester 
1 - https://github.com/espressif/arduino-esp32/tree/master/libraries/SD
2 - de voir si ce reader qui me semble utiliser le pinout MMC est vraiment compatible SPI, et de tester avec un autre reader

ul42, 2022-05-06:
Je vais essayer la librairie que vous me conseiller ;)
Par contre pour ce qui est du pinout MMC, c'est bien le SPI que j'utilise. C'est le même que celui de la Fysetc E4 qui est aussi en SPI et sinon il y aurait usage des broches data dans un ordre de priorité précis me semble t'il. Corriger moi si je raconte une bêtise.

luc-github, 2022-05-06:
ce qu'il faut utiliser c'est https://github.com/espressif/arduino-esp32/blob/master/libraries/SD/examples/SD_Test/SD_Test.ino
il n'y a rien a modifier il fonctionne avec le  pinout sd par default

ul42, 2022-05-06:
Comment dois-je faire pour l'utilliser ?

ul42, 2022-05-08:
Bonjour à vous,
J'ai changé de carte SD, et tout est en fonction.
Je pense que ça doit être le formatage, que je pensais avoir réglé.

Le problème vient de moi.
Je vous remercie malgré tout pour l'aide.
Cordialement.

luc-github, 2022-05-08:
Pour les cartes SD toujours utiliser des 4GB   et les formater avec https://www.sdcard.org/downloads/formatter/

luc-github, 2022-05-08:
Bon devt de carte

closed 2022-05-08 by luc-github