luc-github/ESP3DLib #46: [BUG]failed compilation on ESP401.cpp

nikybiasion, 2022-08-12:
**Describe the bug**
Compilation fail with error:
.pio/libdeps/mks_tinybee/ESP3DLib/src/core/espcmd/ESP401.cpp: In member function 'bool Commands::ESP401(const char*, level_authenticate_type, ESP3DOutput*)':
.pio/libdeps/mks_tinybee/ESP3DLib/src/core/espcmd/ESP401.cpp:89:34: error: ambiguous overload for 'operator=' (operand types are 'String' and 'bool')
                         response = false;
                                  ^

**To Reproduce**
Steps to reproduce the behavior:
1. pio build

**Marlin Firmware:**
- Marlin Branch/Version: 2.0.x MKS Tinybee

**ESP3DLib :**
- ESP3DLib Branch/Version: 3.0

luc-github, 2022-08-12:
Are you sure you are using latest version ? - you may need to do a clear all in platformio/vscode to update latest ESP3DLib code
This issue has been fixed 3months ago :
[image]

And I have just tested compilation pass with AUTHENTICATION_FEATURE 
[image]

Be aware if you update code now you will get several updates and some need also the webUI to be updated, so you will have to update the webUI V3 also

luc-github, 2022-08-14:
@nikybiasion let me know if a clear all - fix your issue , so this ticket can be closed

nikybiasion, 2022-08-14:
ok, i've tried to build mks original firmware with esp3dlib3.0, but it fail. With your Marlin branch everything works fine

luc-github, 2022-08-14:
yes Marlin need several changes, not yet implemented

luc-github, 2022-08-15:
I close issue then

closed 2022-08-15 by luc-github