luc-github/ESP3DLib #48: Hi Luc. I have tried ESP3dLib v3.0 on my custom Board. But file gcode and the folder in which there is a gcode file not showing but if I check directly on the printer. gcode files and folders containing gcode files exist.

luc-github, 2022-11-04:
Hi Luc. I have tried ESP3dLib v3.0 on my custom Board. But file gcode and the folder in which there is a gcode file not showing but if I check directly on the printer. gcode files and folders containing gcode files exist.

and controller fan for cooling main board  when stepper or heater on can't function.
when i used esp3dlib v1 with marlin 2.0.9 not happen problem.
 
[image]
[image]
[image]
[image]
[image]
[image]
[image]

_Originally posted by @agoy507 in https://github.com/luc-github/ESP3DLib/issues/39#issuecomment-1302203984_

luc-github, 2022-11-04:
> controller fan for cooling main board when stepper or heater on can't function.

I do not know your board specs but ESP3DLib does not do anything  with these things check with latest Marlin to see if it is working - may be some configuration /code change - my fork is base on Marlin 2.1

>file gcode and the folder in which there is a gcode file

I see it can list some of content of SD card but not all ? is that correct ?
if Yes what is your SD card capacity ? did you try with another one ? is that formated with  https://www.sdcard.org/downloads/formatter/ ?

luc-github, 2022-11-04:
@agoy507 please continue here

agoy507, 2022-11-04:
Thanks Luc 
I finally got it working!!!!
I follow your advise for formated SDcard with https://www.sdcard.org/downloads/formatter/ ?

luc-github, 2022-11-04:
what do you mean ?

agoy507, 2022-11-05:
I have tried to send notification via telegram, but the message failed to send.  what's wrong?  if via LINE there is no problem.

luc-github, 2022-11-05:
it may be config - did you followed the wiki ?

agoy507, 2022-11-05:
Yes, i follow instruction the wiki.  I filled T1 with telegram token, T2 with channel name, I also tried to fill T2 with botname like  https://github.com/luc-github/ESP3D/issues/578 but still can't work. Any suggestion ?

luc-github, 2022-11-05:
test your settings using your browser
`https://api.telegram.org/bot<your bot token>/sendMessage?chat_id=@<your configured chat room>&text=Hi`

luc-github, 2022-11-10:
@agoy507 did you solved your Issue?

agoy507, 2022-11-12:
hi luc. I'm sorry just update. yesterday i'm busy with my work.

yes i have solved the problem. i forgot to delete "<>" on bot token. thank you for sugestion.
your project very amazing.

now my problem just controller fan not function. now i still collect reference.

luc-github, 2022-11-12:
Ok cool, I am closing issue then
Thank you for the kind words ^_^

closed 2022-11-12 by luc-github