luc-github/ESP3DLib #64: connect to Wifi without password

userosos, 2023-03-04:
Hi! How to connect to Wifi without password? If i set empty password for wifi it can't save the config.

luc-github, 2023-03-05:
what FIrmware version are you using ?
Can you do a screenshot of the empty password cannot save error ?

userosos, 2023-03-06:
```
> what FIrmware version are you using ? Can you do a screenshot of the empty password cannot save error ?

ESP3D Status
Chip ID:  5297
CPU Frequency:  240Mhz
CPU Temperature:  41.7C
Free memory:  165.46 KB
SDK:  v3.2.3-14-gd3e562907
Flash Size:  8.00 MB
Available Size for update:  1.25 MB
Available Size for SPIFFS:  1.31 MB
Baud rate:  250000
Sleep mode:  None
Web port:  80
Hostname:  marlinesp
Active Mode:  Client Station (E8:31:CD:61:B1:14)
Connected to:  Keenetic-3379
Signal:  40%
Phy Mode:  11n
Channel:  5
IP Mode:  DHCP
IP:  192.168.40.77
Gateway:  192.168.40.1
Mask:  255.255.255.0
DNS:  192.168.40.1
Disabled Mode:  Access Point (E8:31:CD:61:B1:15)
FW version:  Marlin 2.1.2 (ESP3D:1.0.11)
WebUI version:  2.1b77
<html><body>
<!--StartFragment--><span class="modal-text">

Browser: |   | Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:109.0) Gecko/20100101 Firefox/110.0
-- | -- | --

<!--EndFragment-->
</body>
</html>
I mean that i goto ESP3D Settings set Station SSID, Station Password is empty. I can see the pic:
![изображение](https://user-images.githubusercontent.com/22872748/223023309-43ca6367-eaba-40f1-982c-1e1dfc33aeb8.png)

```

luc-github, 2023-03-06:
Yes this is a limitation - ESP3DLib V1 and V3 does not support empty password for SSID, neither ESP3D V2.X / V3.0 there is a solution in ESP3D-TFT V1 that is not yet back ported  but it will be only for V3 not older versions

luc-github, 2023-03-06:
Fixed in V3, set as limitation in V1

closed 2023-03-06 by luc-github