luc-github/ESP3DLib #66: Errors when compiling Esp3DLib and Marlin with esp32-s2

JoseAntonioMG, 2023-03-29:
I have designed a 3d printer with Esp32-S2 and have searched for a version of marlin for this purpose, I have found this fork of Marlin:

https://github.com/simon-jouet/Marlin

It works fine, but when incorporating Esp3DLib, compile errors occur, both in version 1.0 and 3.0.

luc-github, 2023-03-29:
ESP3DLib is not compatible with S2 because it works on different core as Marlin, using a wifi on same core as Marlin will result: when steppers are running  to loose steps / hiccups with stepper

Using ESP32 mono core (S2/C3) for your board is bad idea if you want WiFi or BT on your ESP32,  but ok if you use it without.

I had discussion with Simon Jouet about that and he agreed - his version was a POC without WiFi

JoseAntonioMG, 2023-03-30:
I've managed to compile Simon Jouet's version in Arduino IDE with esp32 core 2.0.0:

Target: Esp32s2 dev module
Upload Speed: 921600
USB CDC On Boot: Enabled
USB Firmware MSC On Boot: Disabled
USB DFU On Boot: Disabled
Upload Mode: Internal USB
CPU Frequency: 240mhz
Flash Frequency: 80mhz
Flash Size: 16Mb
Partition Schema: Default 4Mb with spiffs (1.22Mb App/1.5Mb SPIFFS)
Core debug Level: Verbose
PSRAM: Disabled
Port: COM24 (Esp32s2 Dev Module)

I don't want to use WiFi, however I don't get a signal on the USB port, nor on the UART port.

My Espressif board is ESP32-S2 Wrover-1 16Mb Flash

I have made some sample sketches to test the development board and they all work:

Wi-Fi WebServer Example
Example of USB Serial port output
etc etc

Any way how I can compile the source code with a target that allows communication with the board?

JoseAntonioMG, 2023-03-30:
I am also interested in buying Esp32-S3, as long as there is a source version of Marlin for this purpose. I have noticed that you have a branch with this specific project, Marlin to compile it in Esp32-S3, is that correct?

luc-github, 2023-03-30:
My porting for S3 is lacking the I2S support because I am not familiar with it and need to study it, the S3 I2S API is totally different compare to ESP32 and ESP32-S2

About UART I do not remember any issue during my test but there recent issue raised https://github.com/luc-github/ESP3DLib/issues/65

so there is some compatibility issue for sure

that said on ESP3D I have no issue with basic usage of serial, I did not have dig in yet

JoseAntonioMG, 2023-03-30:
I don't use I2S, my development board is simpler and I only use the pins that ESP32 offers. But I would like to know if the source code works on an ESP32-S3, since I would have to redesign my board.

JoseAntonioMG, 2023-03-30:
The pinouts of the ESP32-S2 Wrover and the ESP32-S3 Wroom are different, so I have to redesign everything.

luc-github, 2023-03-30:
The board I have with S3 MCU use I2S,  so I cannot says  if imy porting works without I2S as I did not tested fully

JoseAntonioMG, 2023-03-31:
Ok, thank you very much for the information. I will try Esp32S3 and redesign my board.
I'll tell you how it worked for me.

luc-github, 2023-03-31:
Sure -  I close issue then 
Good luck

closed 2023-03-31 by luc-github