luc-github/ESP3DLib #67: ESP32-S3 reboots on marlin build ESP3DLibV3.0

JoseAntonioMG, 2023-04-07:
### **Description:**
I have successfully compiled the project: "https://github.com/luc-github/Marlin/tree/ESP3DLibV3.0", on an ESP32-S3 module with 16Mb Flash, however the board reboots after recognizing the SD Card.

### **Steps to reproduce the behavior:**
Once the project is compiled and the ESP32-S3 module is flashed, I restart the board and connect the USB to the monitor:

**Monitor Out**

20:01:07.197 > ESP-ROM:esp32s3-20210327
20:01:07.200 > Build:Mar 27 2021
20:01:07.202 > rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
20:01:07.207 > Saved PC:0x403775a0
20:01:07.208 > SPIWP:0xee
20:01:07.210 > mode:DIO, clock div:1
20:01:07.212 > load:0x3fcd0108,len:0x39c
20:01:07.214 > load:0x403b6000,len:0x9a4
20:01:07.216 > load:0x403ba000,len:0x2868
20:01:07.219 > SHA-256 comparison failed:
20:01:07.222 > Calculated: 9127e332554622857d5c753c9ffbb9862bfb72e81ded80db80617d5974000cec
20:01:07.228 > Expected: 867c0d6cb8d08f5a22bebcb1e4a5d457c2700a1c6b76abeb4d778eadcc3446e2
20:01:07.234 > Attempting to boot anyway...
20:01:07.238 > entry 0x403b61c0
20:01:07.505 > start
20:01:07.505 >  Brown out Reset
20:01:07.508 >  Watchdog Reset
20:01:07.508 > Marlin bugfix-2.0.x
20:01:07.510 > echo: Last Updated: 2022-05-17 | Author: (none, default config)
20:01:07.516 > echo: Compiled: Apr  7 2023
20:01:07.518 > echo: Free Memory: 319744  PlannerBufferBytes: 1344
20:01:08.243 > echo:Hardcoded Default Settings Loaded
20:01:08.247 > echo:; Linear Units:
20:01:08.249 > echo:  G21 ; (mm)
20:01:08.250 > echo:; Temperature Units:
20:01:08.253 > echo:  M149 C ; Units in Celsius
20:01:08.255 > echo:; Filament settings (Disabled):
20:01:08.259 > echo:  M200 S0 D1.75
20:01:08.261 > echo:; Steps per unit:
20:01:08.262 > echo:  M92 X80.00 Y80.00 Z400.00 E500.00
20:01:08.266 > echo:; Max feedrates (units/s):
20:01:08.269 > echo:  M203 X300.00 Y300.00 Z5.00 E25.00
20:01:08.272 > echo:; Max Acceleration (units/s2):
20:01:08.275 > echo:  M201 X3000.00 Y3000.00 Z100.00 E10000.00
20:01:08.280 > echo:; Acceleration (units/s2) (P<print-accel> R<retract-accel> T<travel-accel>):
20:01:08.287 > echo:  M204 P3000.00 R3000.00 T3000.00
20:01:08.290 > echo:; Advanced (B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>):
20:01:08.298 > echo:  M205 B20000.00 S0.00 T0.00 J0.01
20:01:08.302 > echo:; Home offset:
20:01:08.303 > echo:  M206 X0.00 Y0.00 Z0.00
20:01:08.306 > echo:; Material heatup parameters:
20:01:08.309 > echo:  M145 S0 H180.00 F0
20:01:08.312 > echo:  M145 S1 H240.00 F0
20:01:08.314 > echo:; Hotend PID:
20:01:08.315 > echo:  M301 P22.20 I1.08 D114.00
20:01:08.318 > echo:; LCD Contrast:
20:01:08.320 > echo:  M250 C255
20:01:13.047 > echo:SD card ok
20:01:22.363 > echo:SD card ok

rebooting ...

20:01:22.387 > ESP-ROM:esp32s3-20210327
20:01:22.391 > Build:Mar 27 2021
20:01:22.392 > rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
20:01:22.397 > Saved PC:0x403775a0
20:01:22.399 > SPIWP:0xee
20:01:22.399 > mode:DIO, clock div:1
20:01:22.401 > load:0x3fcd0108,len:0x39c
20:01:22.404 > load:0x403b6000,len:0x9a4
20:01:22.407 > load:0x403ba000,len:0x2868
20:01:22.409 > SHA-256 comparison failed:
20:01:22.411 > Calculated: 9127e332554622857d5c753c9ffbb9862bfb72e81ded80db80617d5974000cec
20:01:22.418 > Expected: 867c0d6cb8d08f5a22bebcb1e4a5d457c2700a1c6b76abeb4d778eadcc3446e2
20:01:22.425 > Attempting to boot anyway...
20:01:22.427 > entry 0x403b61c0
20:01:22.695 > start
20:01:22.696 >  Brown out Reset
20:01:22.697 >  Watchdog Reset
20:01:22.699 > Marlin bugfix-2.0.x
20:01:22.700 > echo: Last Updated: 2022-05-17 | Author: (none, default config)
20:01:22.706 > echo: Compiled: Apr  7 2023
20:01:22.708 > echo: Free Memory: 319744  PlannerBufferBytes: 1344
20:01:23.434 > echo:Hardcoded Default Settings Loaded
20:01:23.437 > echo:; Linear Units:
20:01:23.439 > echo:  G21 ; (mm)
20:01:23.441 > echo:; Temperature Units:
20:01:23.443 > echo:  M149 C ; Units in Celsius
20:01:23.446 > echo:; Filament settings (Disabled):
20:01:23.449 > echo:  M200 S0 D1.75
20:01:23.451 > echo:; Steps per unit:
20:01:23.453 > echo:  M92 X80.00 Y80.00 Z400.00 E500.00
20:01:23.457 > echo:; Max feedrates (units/s):
20:01:23.459 > echo:  M203 X300.00 Y300.00 Z5.00 E25.00
20:01:23.464 > echo:; Max Acceleration (units/s2):
20:01:23.466 > echo:  M201 X3000.00 Y3000.00 Z100.00 E10000.00
20:01:23.470 > echo:; Acceleration (units/s2) (P<print-accel> R<retract-accel> T<travel-accel>):
20:01:23.477 > echo:  M204 P3000.00 R3000.00 T3000.00
20:01:23.481 > echo:; Advanced (B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>):
20:01:23.489 > echo:  M205 B20000.00 S0.00 T0.00 J0.01
20:01:23.492 > echo:; Home offset:
20:01:23.494 > echo:  M206 X0.00 Y0.00 Z0.00
20:01:23.496 > echo:; Material heatup parameters:
20:01:23.500 > echo:  M145 S0 H180.00 F0
20:01:23.502 > echo:  M145 S1 H240.00 F0
20:01:23.504 > echo:; Hotend PID:
20:01:23.506 > echo:  M301 P22.20 I1.08 D114.00
20:01:23.509 > echo:; LCD Contrast:
20:01:23.510 > echo:  M250 C255
20:01:28.258 > echo:SD card ok
20:01:37.554 > echo:SD card ok

rebooting ...

20:01:37.578 > ESP-ROM:esp32s3-20210327
20:01:37.581 > Build:Mar 27 2021
20:01:37.583 > rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
20:01:37.590 > Saved PC:0x403775a0
20:01:37.590 > SPIWP:0xee
20:01:37.592 > mode:DIO, clock div:1
20:01:37.592 > load:0x3fcd0108,len:0x39c
20:01:37.594 > load:0x403b6000,len:0x9a4
20:01:37.596 > load:0x403ba000,len:0x2868
20:01:37.600 > SHA-256 comparison failed:
20:01:37.602 > Calculated: 9127e332554622857d5c753c9ffbb9862bfb72e81ded80db80617d5974000cec
20:01:37.608 > Expected: 867c0d6cb8d08f5a22bebcb1e4a5d457c2700a1c6b76abeb4d778eadcc3446e2
20:01:37.615 > Attempting to boot anyway...
20:01:37.618 > entry 0x403b61c0
20:01:37.885 > start
20:01:37.886 >  Brown out Reset
20:01:37.888 >  Watchdog Reset
20:01:37.889 > Marlin bugfix-2.0.x
20:01:37.891 > echo: Last Updated: 2022-05-17 | Author: (none, default config)
20:01:37.896 > echo: Compiled: Apr  7 2023
20:01:37.899 > echo: Free Memory: 319744  PlannerBufferBytes: 1344
20:01:38.624 > echo:Hardcoded Default Settings Loaded
20:01:38.628 > echo:; Linear Units:
20:01:38.630 > echo:  G21 ; (mm)
20:01:38.631 > echo:; Temperature Units:
20:01:38.633 > echo:  M149 C ; Units in Celsius
20:01:38.637 > echo:; Filament settings (Disabled):
20:01:38.640 > echo:  M200 S0 D1.75
20:01:38.641 > echo:; Steps per unit:
20:01:38.643 > echo:  M92 X80.00 Y80.00 Z400.00 E500.00
20:01:38.646 > echo:; Max feedrates (units/s):
20:01:38.649 > echo:  M203 X300.00 Y300.00 Z5.00 E25.00
20:01:38.654 > echo:; Max Acceleration (units/s2):
20:01:38.656 > echo:  M201 X3000.00 Y3000.00 Z100.00 E10000.00
20:01:38.660 > echo:; Acceleration (units/s2) (P<print-accel> R<retract-accel> T<travel-accel>):
20:01:38.669 > echo:  M204 P3000.00 R3000.00 T3000.00
20:01:38.672 > echo:; Advanced (B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>):
20:01:38.680 > echo:  M205 B20000.00 S0.00 T0.00 J0.01
20:01:38.684 > echo:; Home offset:
20:01:38.684 > echo:  M206 X0.00 Y0.00 Z0.00
20:01:38.687 > echo:; Material heatup parameters:
20:01:38.691 > echo:  M145 S0 H180.00 F0
20:01:38.693 > echo:  M145 S1 H240.00 F0
20:01:38.695 > echo:; Hotend PID:
20:01:38.696 > echo:  M301 P22.20 I1.08 D114.00
20:01:38.699 > echo:; LCD Contrast:
20:01:38.701 > echo:  M250 C255
20:01:43.464 > echo:SD card ok
20:01:52.744 > echo:SD card ok

rebooting ...

### **Marlin Firmware:**
- Marlin bugfix-2.0.x

### **ESP3DLib :**
- ESP3DLib=https://github.com/luc-github/ESP3DLib#3.0
- arduinoWebSockets=https://github.com/links2004/arduinoWebSockets
- ESP32SSDP=https://github.com/luc-github/ESP32SSDP
- SDFat2=greiman/SdFat@2.1.2

### **Board used (please complete the following information):**
 - BOARD: ESP32-S3
 - Flash size: 16Mb
 - SD Card Reader SPI
 - HotEnd
 - Driver A4988
 - Fan HotEnd
 - End Stops (X, Y and Z)

luc-github, 2023-04-08:
Sorry this branch is not up to date and the S3 part was not working due to absence of porting of i2S for S3 
So I forget the state - the reboot is most likely due I2S enabled or to some debug code to work around it

JoseAntonioMG, 2023-04-08:
In the pins_MKS_TINYBEE_V2.h file I have disabled the I2S option:

//
// Disable I2S stepper stream
//
#undef I2S_STEPPER_STREAM
#define I2S_WS                                -1
#define I2S_BCK                               -1
#define I2S_DATA                             -1
#undef LIN_ADVANCE                      // Currently, I2S stream does not work with linear advance

I guess this way, the I2S part of the code won't be running.

Is it possible that one day you can resume code debugging for S3? It would be very interesting, since this type of board allows you to simplify the design of controllers for 3D printers, better than the ESP32 does.

luc-github, 2023-04-08:
As you can see in latest updated btanch (https://github.com/luc-github/Marlin/tree/ESP3D-V3-2.1.x) which is no more up to date FYI the S3 has been removed because I have no bandwidth/motivation for supporting Marlin 
So this `one day` it is unlikely in close timeframe sorry

JoseAntonioMG, 2023-04-08:
On the other hand Luc, is there another branch of Marlin for ESP32-S3? what do you know? I can't find it on the official Marlin page, however some developer may have implemented it

luc-github, 2023-04-08:
Not I am aware of

JoseAntonioMG, 2023-04-08:
Thank you for your answers Luc, I hope that one day you will resume this part of the development of Marlin for ESP32-S3

JoseAntonioMG, 2023-04-08:
I'll be waiting ...

luc-github, 2023-04-08:
I close issue then - good luck with your project

closed 2023-04-08 by luc-github