luc-github/ESP3DLib #72: Some help needed on how to install the WEB interface.

JazzinSA, 2023-10-05:
I managed to get a clean compile without any errors or warnings. However, when booting the machine, I can log onto the machine and I get the landing page but cannot find the upload option to install the ESP3D Web UI like with version 1.0. Would love some pointers here.  of how to proceed.

luc-github, 2023-10-05:
can you share the screenshot of the page you get ?

JazzinSA, 2023-10-05:
[image]

luc-github, 2023-10-05:
here
[image]

luc-github, 2023-10-05:
and the file is this one : https://github.com/luc-github/ESP3D-WEBUI/tree/3.0/dist/Printer3D/Marlin-embedded

JazzinSA, 2023-10-05:
Sorted, Thank you,

closed 2023-10-05 by JazzinSA