luc-github/ESP3DLib #75: [BUG NOTE]Marlin bugfix 2.1.x is broken for ESP3DLib 1.0

luc-github, 2023-12-15:
https://github.com/MarlinFirmware/Marlin/commit/fecadaca82e49c2558b42684382ede121274bfea removed all flag for ESP3DLib ###Configuration_adv.h
Prevent  ESP3D_WIFISUPPORT  to use 
```
#define WEBSUPPORT          // Start a webserver (which may include auto-discovery) using SPIFFS
#define OTASUPPORT          // Support over-the-air firmware updates
#define WIFI_CUSTOM_COMMAND // Accept feature config commands (e.g., WiFi ESP3D) from the host
```

You need to replace:
`#if ENABLED(WIFISUPPORT)` by `#if ANY(WIFISUPPORT, ESP3D_WIFISUPPORT)`

Or add the flags outside of the `#if ENABLED(WIFISUPPORT)`

If not, ESP3DLib cannot communicate with Marlin and is not setup for web access

luc-github, 2024-02-24:
https://github.com/MarlinFirmware/Marlin/issues/26786#issuecomment-1961813818

thisiskeithb, 2024-03-22:
Hi Luc! Sorry about that.

`2.1.2.2` may not get retagged with fixes, but hopefully my PR (https://github.com/MarlinFirmware/Marlin/pull/26822) will get merged into `bugfix-2.1.x` soon and it'll restore ESP3D support.

luc-github, 2024-03-22:
Hi @thisiskeithb thank you, no worry
thank you for your help

closed 2024-12-09 by luc-github