luc-github/ESP3DLib #77: Compilation Error for POLAR 3D PRINTER in Marlin-ESP3DLib-V3-bugfix-2.1.x

Gokuh1980, 2024-01-12:
**Describe the bug**
When i setup Vscode to compile for FYSETC E4 a polar 3d Printer it aborts due to several warnings
I Have a NODEMCU ESP32S but they have the same pins so i used FYSETC E4

[image]

[image]

**Marlin Firmware:**
- Marlin Branch/Version:[e.g bugfix-2.1.x] 
- Kinetamits changed to POLAR  

**ESP3DLib :**
- ESP3DLib Branch/Version: 3  
-  
- Wifi mode:AP
 - BOARD: ESP32 FYSETC 4 
 - Flash size: 16M

luc-github, 2024-01-12:
What Marlin repository are you using ? And what branch ?

Gokuh1980, 2024-01-12:
Hi! thanks for your response!

https://github.com/luc-github/Marlin/tree/ESP3DLib-V3-bugfix-2.1.x
[image]

luc-github, 2024-01-12:
the issue is not linked to ESP3DLib
check if you have same issue with latest Marlin Buxfix 2.1 without enabling wifi because upstream is broken

luc-github, 2024-01-14:
@Gokuh1980 any update ? can you compile with Upstream Marlin ?

Gokuh1980, 2024-01-14:
Hi! Yes, i tried to compile the bugfix from marlin today and it fails with similar errors.

luc-github, 2024-01-14:
does latest release having same issue ? https://github.com/MarlinFirmware/Marlin/releases/tag/2.1.2.1

Gokuh1980, 2024-01-14:
Yes, i've tried it and it compiles ok, but that release doesn't have the polar geometry i'm trying to build.

Algo i really really like your work with esp3d and wifi.

So i think i will wait a little longer.

(I'm no so good at coding)

luc-github, 2024-01-14:
ho then I suggest you to push issue in Marlin and follow resolution - then update here when fixed - I will then update the fork

Gokuh1980, 2024-01-14:
Yes of course!!! Thank you!!! Keep up with your project, its fantastic!

luc-github, 2024-01-14:
Thank you - sorry I have no bandwidth to follow all Marlin update so your help is really appreciate

luc-github, 2024-01-15:
no but my fork was not up to date - I have just did a rebase from bugfix now- so you can give a try 
Be noted the fix you mention is not present yet so you need to apply it manually

luc-github, 2024-01-16:
the warnings are normal and harmless

The config you provided is for ESP32 not FYSETC_E4 you must use FYSETC_E4  board if this is what you have

In both case ESP32 / FYSETC_E4 I cannot duplicate using your files 

[image]

luc-github, 2024-01-20:
@Gokuh1980 any update ?

Gokuh1980, 2024-01-20:
hi Luc! i'm trying to compile in another computer at home with ubuntu but keep getting this error...

[image]

strange is that i don't have any display configured...  

[Marlin.zip](https://github.com/luc-github/ESP3DLib/files/13998231/Marlin.zip)

luc-github, 2024-01-21:
>strange is that i don't have any display configured...

Really ?

But your configuration.h has :

#define MKS_MINI_12864_V3

and 
#define LCD_INFO_SCREEN_STYLE 0

and I just did a simple windiff with my repo

luc-github, 2024-01-22:
sorry Idea for what? I do not understand the quuestion

Gokuh1980, 2024-01-22:
i was having an error on upload because fysetc_e4 as a flash of 16mb and my board has 4mb, so it results in a bootloop. thats why i wanted to ask you if you have some testing compiling for a 4mb esp32. 

now i have changed esp32.csv partition file and it boots:

[image]

now i have to resolve the errors in the boot log.

luc-github, 2024-01-22:
you just need to edit the env and adjust board size and use proper partition like min-spiffs one

closed 2024-01-22 by luc-github