luc-github/ESP3DLib #80: [Question] Do i need to do anything other than enabling?

ahakkoca07, 2024-11-04:
What is your question ? 
Do i need to do anything other than enabling ESP3DLib in Configuration_adv.h?

The relevant code in Configuration_adv.h:
```

/**
 * Native ESP32 board with WiFi or add-on ESP32 WiFi-101 module
 */
//#define WIFISUPPORT         // Marlin embedded WiFi management. Not needed for simple WiFi serial port.
#define ESP3D_WIFISUPPORT   // ESP3D Library WiFi management (https://github.com/luc-github/ESP3DLib)

/**
 * Extras for an ESP32-based motherboard with WIFISUPPORT
 * These options don't apply to add-on WiFi modules based on ESP32 WiFi101.
 */
#if ANY(WIFISUPPORT, ESP3D_WIFISUPPORT)
  #define WEBSUPPORT          // Start a webserver (which may include auto-discovery) using SPIFFS
  #define OTASUPPORT          // Support over-the-air firmware updates
  #define WIFI_CUSTOM_COMMAND // Accept feature config commands (e.g., WiFi ESP3D) from the host

```
Do i still have to enable SERIAL_PORT_2 like seen [here](https://github.com/luc-github/ESP3DLib?tab=readme-ov-file#how-to-enable-) , if yes, do i have to connect anything to the said pins?

luc-github, 2024-11-04:
please read Documenation according your version

https://esp3d.io/ESP3DLib/

ahakkoca07, 2024-11-04:
Thank you!
Also; for your information, some links in Documentation is broken, like the 3.0 Tested boards:

> Tested boards
> The list is [here](https://esp3d.io/ESP3DLib/Version_1.X/hardware/) for 1.0, and [here](https://esp3d.io/ESP3DLib/Version_3.x/hardware/) for 3.0
> 
> Please refer to Marlin documentation for more information on specific configurations for each boards.

ahakkoca07, 2024-11-04:
No problem!
Can i just define 2 unused pins of the esp32 as SERIAL_PORT_2 ?

luc-github, 2024-11-04:
where do you see you need to define pins for SERIAL_PORT_2  ?

ahakkoca07, 2024-11-04:
Oh sorry, my bad. I thought we were defining the pins here, but it wasn't the case, thank you.

```
Configuration.h
Select an ESP32 based board.

Uncomment the second serial port to allow esp3d to get all printer feedback

/**
 * Select a secondary serial port on the board to use for communication with the host.
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT_2 -1
```

closed 2024-11-04 by luc-github