luc-github/ESP3DLib #81: [BUG] MarlinConfigPre.h: No such file or directory

ahakkoca07, 2024-11-05:
**Describe the bug**
Arduino IDE Compilation error: exit status 1
```
In file included from c:\Users\User\Documents\GitHub\libraries\ESP3DLib\src\command.cpp:21:
c:\Users\User\Documents\GitHub\libraries\ESP3DLib\src\esp3dlibconfig.h:45:43: fatal error: ../../../../../Marlin/src/inc/MarlinConfigPre.h: No such file or directory
   45 | #include MARLIN_PATH(inc/MarlinConfigPre.h)
      |                                           ^
compilation terminated.
exit status 1

Compilation error: exit status 1

```
**To Reproduce**
Steps to reproduce the behavior:
1. Open Marlin.ino in Arduino IDE
2. Click on "Verify/Compile"
3. Wait about 60 minutes for it to compile
4. See error

**Expected behavior**
Compilation without error

**Screenshots**
[image]

**Marlin Firmware:**
- Marlin Branch/Version:[2.1.2.4] 
- Any change done in Marlin code/configuration: Added new board.

**ESP3DLib :**
- ESP3DLib Branch/Version:[master/[4fbed7f](https://github.com/luc-github/ESP3DLib/commit/4fbed7fefa48df06c912b9604a1a56255cfe1890)]  
- Web UI Version:[e.g 2.1b14]  
- Wifi mode:[e.g. STA]  
- Any change done in ESP3DLib code/configuration: No.

**Board used (please complete the following information):**
 - BOARD: [Lolin S2 Mini] 
 - Flash size: [4M]
 - partitions: [default]

**Additional context**
Add any other context about the problem here.
Marlin.ino location: ../Documents/Marlin/Marlin-2.1.2.4/Marlin/Marlin.ino
ESP3DLib location: ../Documents/GitHub/libraries/ESP3DLib
I installed ESP3DLib like this, but i don't know if it's relevant.
1. Clicked "Download ZIP" from the Github page.
2. In Arduino IDE: "Sketch/Include Library/Add .ZIP Library..." and selected the downloaded zip file.

luc-github, 2024-11-05:
ESP3DLib 1.0 only support platformio
ESP3DLib 3.0 support arduino by adding define

ahakkoca07, 2024-11-05:
Hmm, how/where can i add the define?
Also, which branch i should select? The "3.0" one?
[image]

luc-github, 2024-11-05:
for 3.0 is not that simple please read https://esp3d.io/ESP3DLib/

need my fork of Marlin as mentionned in doc/readme

the define is here : https://github.com/luc-github/ESP3DLib/blob/3.0/src/esp3dlib_config.h#L25

ahakkoca07, 2024-11-05:
Hi again, just to make sure, i will use this Marlin: https://github.com/luc-github/Marlin
And i will use this ESP3DLib: https://github.com/luc-github/ESP3DLib/tree/3.0
I will add the define here: https://github.com/luc-github/ESP3DLib/blob/3.0/src/esp3dlib_config.h#L25
Is this correct?
Also, can i use the 2.1.2.4 config files in your 2.1.2 version of marlin?

ahakkoca07, 2024-11-05:
Thank you for your help!
I will compare the files right now.

luc-github, 2024-11-05:
what board are you using ?

ahakkoca07, 2024-11-05:
Yes, ESP32-S2 Mini

ahakkoca07, 2024-11-05:
> what board are you using ? a lolin esp32 S2? S2 is not supported in Marlin
But MRR_ESPA board is using an ESP32-S2 and is supported

luc-github, 2024-11-05:
What do you expect to do ?
S2 is not supported by Marlin

luc-github, 2024-11-05:
no MRR_ESPA use ESP32

luc-github, 2024-11-05:
use single is not tested neither recommended for esp3dlib for sure

ahakkoca07, 2024-11-05:
Sorry, i mixed it with [this](https://github.com/simon-jouet/ESP32Controller)

ahakkoca07, 2024-11-05:
> use single core is not tested neither recommended for esp3dlib for sure because marlin use one core and wifi/esp3d use one core

Hmm, can i use a ESP32-WROOM-32D then?

luc-github, 2024-11-05:
this board was.a.proof of concept and not planned to.use.wifi

any esp32 should fit

ahakkoca07, 2024-11-05:
Thank you, i will use an ESP32-WROOM-32D then.

ahakkoca07, 2024-11-05:
[image]
https://lastminuteengineers.com/esp32-wroom-32-pinout-reference/
This seems like it will be _a bit_ hard lol

ahakkoca07, 2024-11-05:
[image]
Uh, i only have one pin left and i have to choose one of the following:

```
//#define BEEPER_PIN                               // Buzzer
//#define PS_ON_PIN                                // Power Supply
//#define FIL_RUNOUT_PIN                           // Filament Runout Sensor
//#define NEOPIXEL_PIN                             // Neopixels
//#define SD_DETECT_PIN                            // Sd card detection pin

```
Which choice would be best? What do you think? I also want to know if i can use this as a I/O expander since it uses the pcf8574 chip too.
[image]

luc-github, 2024-11-05:
I close ticket as answered

closed 2024-11-05 by luc-github