luc-github/ESP3DLib #82: enable 3 serial connexion : {USB, MKS TFT24, WIFI}

moriamoria, 2024-12-15:
Hello,

I like to use 3 connexions for tinybee motherboard (USB, MKS TFT24, WIFI).

I have been able to configure and uses 2 connexions (USB, MKS TFT24)

By the way, I had to invert mkstft24 Tx and Rx, as they are not in the same pin order.

But when I activate ESP3D_WIFISUPPORT, i loose the mks tft24 connexion.

I supose that esp3Dlib overwrite the port number 2. But that just an asumption.

I have test with Marlin bugfix 2.1 and native mks tnybee firmware 2.0.9.2.

Can you help on this?

Regards

luc-github, 2024-12-16:
My  Marlin fork for esp3dlib 3.0  (https://github.com/luc-github/Marlin/tree/ESP3DLib-V3-bugfix-2.1.x) has this feature 

here example of configuration : https://github.com/luc-github/MarlinConfigurations/tree/bugfix-2.1.x/config/examples/Mks/TinyBeeV1/MKS-SerialTFT

moriamoria, 2024-12-16:
Thanks,

I will do the test.

but the configuration looks strange, as in a previous post, "elensp" clearly told be not to use -1 for serial port : 

```
no, you sequentially enable the number if serial ports you need
SERIAL_PORT, SERIAL_PORT_2 then SERIAL_PORT_3
you assign what ever uart you want to use. to any enabled serial_port (there are some exceptions to this)

Your board has two uarts:
0 is the USB cable
2 is on the exp port for serial displays

-1 is not valid, It is for chips with inbuilt usb in hardware, this board uses a ch340

All you need is
#define SERIAL_PORT 0
and comment out #define SERIAL_PORT_2
```

—

I will tell you today my test.

many thanks

luc-github, 2024-12-16:
what branch did you use?

moriamoria, 2024-12-16:
hello,

Update features.ini
origin/ESP3DLib-V3-2.1.2
23/03/2024 01:49:34

regards

moriamoria, 2024-12-16:
Hello,

I got some result by mofifying the integration of your library inside Marlin code.

I really don't know if it is a valid correction, or if I break some coding rules.

my understanding is that we overload MYSERIAL2 with  : 

    typedef ForwardSerial1Class< decltype(Serial2Socket) > DefaultSerial1;
    extern DefaultSerial1 MSerial0;
    #define MYSERIAL2 MSerial0

So I have just create a new instance of MYSERIAL3 and flushableSerial2

    #define MYSERIAL3 flushableSerial2

I post my modified source code for your review if you can. 
[ESP32.zip](https://gi
[MarlinCore.zip](https://github.com/user-attachments/files/18158135/MarlinCore.zip)
thub.com/user-attachments/files/18
[Marlin.zip](https://github.com/user-attachments/files/18158136/Marlin.zip)
158130/ESP32.zip)

For the moment I failed to do the same modification on the last bugfix 2.1.

I wait for your comments.

Many thanks for the support

Regards

luc-github, 2024-12-17:
> Hello,
> 
> I forgot the origin :
> 
> https://github.com/luc-github/Marlin.git
> 
> Regards

I means you did not follow my link and used the wrong code - so it is normal it is not working

closed 2024-12-17 by luc-github