luc-github/ESP3DLib #85: [Question] how Controlling esp32 with wifi serial using slicers

Ammar56788, 2025-03-04:
Hello;
I have simple esp32(30pin devkit v1) as main board controller for 3dprinter, and I enabled esp3d.
I can use browser to control 3d printer, and upload gcode etc...

But if I want to connect it with slicer over wifi, and send data with wifi gcode, I can't.
I think maybe there are disabled feature that I don't know, or maybe I must use wifi support instead of esp3d.

Second: I don't know esp3d and wifisupport difference in marlin...

Third : Also when processing the gcode on the SD card by esp32 I can't send any command by esp3d, it gives an error: busy.... What to do to control the flow speed and feed rate?

i try to use esp3dlib v3.x in marlin2.1.2.4(changing ini/feature.ini esp3dlib's link) but it error

[Configuration.txt](https://github.com/user-attachments/files/19077205/Configuration.txt)

 when compiling.

but Someone suggested me: "What your looking for is "Serial/Wifi bridge using configurable port 8888" part of esp3d"

but I don't know what this is and what it has to do with this issue.

Configuration.txt  and  Configuration_adv.txt:

[Configuration.txt](https://github.com/user-attachments/files/19077213/Configuration.txt)
[Configuration_adv.txt](https://github.com/user-attachments/files/19077221/Configuration_adv.txt)

Ammar56788, 2025-03-04:
i dont know why esp32(30pin devkit v1) not supported by esp3dlib v3.x. it has same core hardware with another board... :///

luc-github, 2025-03-05:
>if I want to connect it with slicer over wifi, and send data with wifi gcode, I can't.
I think maybe there are disabled feature that I don't know, or maybe I must use wifi support instead of esp3d.

I am guessing your using Marlin with esp3dlib 1.0 , as you see here https://esp3d.io/ESP3DLib/Version_1.X/features/ telnet/raw tcp is not a feature so it is normal you cannot acheive what you want - you need ESP3DLib 3.0 https://esp3d.io/ESP3DLib/Version_3.X/features/

>Second: I don't know esp3d and wifisupport difference in marlin...

Marlin has very basic support in Marlin, ESP3DLib has more features

>Third : Also when processing the gcode on the SD card by esp32 I can't send any command by esp3d, it gives an error: busy.... What to do to control the flow speed and feed rate?

Busy come from Marlin itself, if heating it do not take any new command until temperature is reached - without context I cannot give more answer, ESP3DLib is just a wifi bridge.

>i try to use esp3dlib v3.x in marlin2.1.2.4(changing ini/feature.ini esp3dlib's link) but it error when compiling.

Yest that is normal, for ESP3DLib 3.0 as mentionned in readme https://github.com/luc-github/ESP3DLib/tree/3.0  and documentation https://esp3d.io/ESP3DLib/Version_3.X/, currently it need a fork of Marlin because I am still working on it and changes are not yet pushed to Marlin, it also need to update webui as mentionned

>but Someone suggested me: "What your looking for is "Serial/Wifi bridge using configurable port 8888" part of esp3d"

Yes as mentionned above that is in ESP3DLIB 3.0, be aware that ESP3D (https://esp3d.io/ESP3D/) is different than ESP3DLib, it is for ESP daughter board not main board, and it has more feature in 1.0/2.0 than ESP3DLib, in 3.0 they are using exactly same code base.

>i dont know why esp32(30pin devkit v1) not supported by esp3dlib v3.x. it has same core hardware with another board... :///

ESP3DLib do not care the ESP board but need to be ESP32 dual core, you issue is you just change version in ini file but it is not enough that is why currently need a Marlin fork

Ammar56788, 2025-03-05:
special version of Marlin not synced with latest version of Marlin that I work with it. And configuration.h that personalized...

By the way, Disclaimer content make me afraid of I using ESP3DLib 3.0 for diy custom ESP32 3dprinter controller... that when working with this version, the board will catch fire or... :))))

And what change I need to do in configuration.h ??

Ammar56788, 2025-03-05:
This issue will be closed as soon, but if there was another question, another issue will be born...😐

luc-github, 2025-03-05:
>special version of Marlin not synced with latest version of Marlin that I work with it. And configuration.h that personalized...

Yes I am late 

>By the way, Disclaimer content make me afraid of I using ESP3DLib 3.0 for diy custom ESP32 3dprinter controller... that when working with this version, the board will catch fire or... :))))

Of course - I do not want to be responsible of anything - so it is normal warning

>And what change I need to do in configuration.h ??

Current configuration is using ESP3DLib V3 with tinybee board so just change what is related to your board

luc-github, 2025-03-05:
may I know what board you are using ? esp32(30pin devkit v1)  is not a board for 3d printing it is eventually a core board for a ramp like board

Ammar56788, 2025-03-05:
Esp32 devkit v1 with ramps 1.5 (using diy pin converter)

luc-github, 2025-03-05:
Ok thank you - I was curious

closed 2025-03-05 by Ammar56788