luc-github/ESP3DLib #86: [BUG] soc/rtc_wdt.h not found error

ahakkoca07, 2025-03-05:
**Describe the bug**
A clear and concise description of what the bug is.
#include <soc/rtc_wdt.h> line gives error.

**To Reproduce**
Compile Marlin fork of Luc with ESP32 Dev Board and ESP32Lib 3.0.

**Expected behavior**
Compile without error.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Marlin Firmware:**
- Marlin Branch/Version:[635d22c](https://github.com/luc-github/Marlin/commit/635d22c1e1e6b1be1d8e065aaf35dd457243c8f8)
- Any change done in Marlin code/configuration:
Added custom board

**ESP3DLib :**
- ESP3DLib Branch/Version:[b5c99e2](https://github.com/luc-github/ESP3DLib/commit/b5c99e2ca8d089660ae588b521f5fe4f55f24373)
- Web UI Version:unknown
- Wifi mode:unknown
- Any change done in ESP3DLib code/configuration: Enabled Arduino IDE as advised in [#81](https://github.com/luc-github/ESP3DLib/issues/81#issuecomment-2456849499)
**Board used (please complete the following information):**
 - BOARD: ESP32 Dev Board
 - Flash size: 4M
 - partitions: default

**Additional context**
```
Arduino:1.8.19 (Linux), Kart:"ESP32 Dev Module, Disabled, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, Core 1, Core 1, None, Disabled, Disabled"

/home/ahakkoca07/Belgeler/arduino-1.8.19/libraries/ESP3DLib/src/core/hal.cpp:30:10: fatal error: soc/rtc_wdt.h: No such file or directory
   30 | #include <soc/rtc_wdt.h>
      |          ^~~~~~~~~~~~~~~
compilation terminated.
"WiFi.h" için birden fazla library bulundu
Kullanılıyor: /home/ahakkoca07/.arduino15/packages/esp32/hardware/esp32/3.1.3/libraries/WiFi
Kullanılmıyor: /home/ahakkoca07/Belgeler/arduino-1.8.19/libraries/WiFi
exit status 1
ESP32 Dev Module kartı için derleme hatası.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

I think we can solve this like the following: https://github.com/espressif/arduino-esp32/discussions/9800
I currently tried removing the "soc/" from the #include <soc/rtc_wdt.h>, now it is compiling again. I will update when it complies/gives error again.

Well, as i was about to post this, it gave another error:
```
/home/ahakkoca07/Belgeler/arduino-1.8.19/libraries/ESP3DLib/src/esp3dlib.cpp:34:10: fatal error: soc/rtc_wdt.h: No such file or directory
   34 | #include <soc/rtc_wdt.h>
      |          ^~~~~~~~~~~~~~~
```
It got past the previous error, so i will create a PR about this once i find all the errors.

ahakkoca07, 2025-03-05:
Okay, i ran compile again.

[log.log](https://github.com/user-attachments/files/19097716/log.log)

luc-github, 2025-03-05:
no I take no pr on  my marlin fork neither esp3dlib if issue cannot be duplicate

you give.me a commit that is not a  branch that is supported which compile fine on platformio, , just give me the branch 

you do not give clearly the change you did in code,  the settings you apply in esp3dlib, please provide  complete list of changes to let me reproduce

you do not give the esp32.arduino core version you are using which seems actually your issue , please provide it

ahakkoca07, 2025-03-06:
ESP3DLib branch is 3.0, sorry, i thought commits were connected to branches. I am still new to GitHub.
The Marlin branch is ESP3DLib-V3-2.1.2(Latest).
In ESP3DLib i only defined the Arduino IDE line (COMPILE_WITH_ARDUINO_IDE).
In Marlin i added a custom board pins file, and enabled the OLED screen.
Arduino:1.8.19 (Linux)(Latest), ESP32 Arduino Board: 3.1.3(Latest)

luc-github, 2025-03-06:
>ESP3DLib branch is 3.0, sorry, i thought commits were connected to branches. I am still new to GitHub.

Yes but I have so may branches that searching the commit is waste of time

>Arduino:1.8.19 (Linux)(Latest Legacy), ESP32 Arduino Board: 3.1.3(Latest)

that the problem - Marlin does not support that version of the core yet - I work on it - you must use the esp32 core version 2.0.7,  the 3.X is a breaking change and Marlin need a lot of update and I need time to focus on lt

ahakkoca07, 2025-03-06:
> you must use the esp32 core version 2.0.7
Thanks, i tried installing that version and so far it is compiling, i will give updates once it finishes.

ahakkoca07, 2025-03-06:
Update: It passed the previous checks successfully, but now it got stuck here:
```
In file included from /home/ahakkoca07/Belgeler/GitHub/Marlin/Marlin/src/HAL/shared/Marduino.h:36,
                 from /home/ahakkoca07/Belgeler/GitHub/Marlin/Marlin/src/HAL/AVR/u8g_com_HAL_AVR_sw_spi.cpp:62:
/home/ahakkoca07/Belgeler/GitHub/Marlin/Marlin/src/HAL/AVR/u8g_com_HAL_AVR_sw_spi.cpp: In function 'void u8g_com_arduino_init_shift_out(uint8_t, uint8_t)':
/home/ahakkoca07/.arduino15/packages/esp32/hardware/esp32/2.0.7/cores/esp32/Arduino.h:120:93: error: cannot convert 'volatile uint32_t*' {aka 'volatile unsigned int*'} to 'volatile uint8_t*' {aka 'volatile unsigned char*'} in assignment
 #define portOutputRegister(port)    ((volatile uint32_t*)((port)?GPIO_OUT1_REG:GPIO_OUT_REG))
                                                                                             ^
/home/ahakkoca07/Belgeler/GitHub/Marlin/Marlin/src/HAL/AVR/u8g_com_HAL_AVR_sw_spi.cpp:71:17: note: in expansion of macro 'portOutputRegister'
   u8g_outData = portOutputRegister(digitalPinToPort(dataPin));
                 ^~~~~~~~~~~~~~~~~~
/home/ahakkoca07/.arduino15/packages/esp32/hardware/esp32/2.0.7/cores/esp32/Arduino.h:120:93: error: cannot convert 'volatile uint32_t*' {aka 'volatile unsigned int*'} to 'volatile uint8_t*' {aka 'volatile unsigned char*'} in assignment
 #define portOutputRegister(port)    ((volatile uint32_t*)((port)?GPIO_OUT1_REG:GPIO_OUT_REG))
                                                                                             ^
/home/ahakkoca07/Belgeler/GitHub/Marlin/Marlin/src/HAL/AVR/u8g_com_HAL_AVR_sw_spi.cpp:72:18: note: in expansion of macro 'portOutputRegister'
   u8g_outClock = portOutputRegister(digitalPinToPort(clockPin));
                  ^~~~~~~~~~~~~~~~~~
Multiple libraries were found for "WiFi.h"
 Used: /home/ahakkoca07/.arduino15/packages/esp32/hardware/esp32/2.0.7/libraries/WiFi
 Not used: /home/ahakkoca07/Belgeler/arduino-1.8.19/libraries/WiFi
exit status 1
Error compiling for board ESP32 Dev Module.
```
I will try downgrading U8glib-HAL now.

luc-github, 2025-03-06:
you need to use same version of librariies defined in platformio ini files
you better use vscode and  platformio - everything is automaticaly setup

ahakkoca07, 2025-03-06:
I tried platformio on windows before, and it was a nightmare for me. But i am currently on linux and i will try it again, thanks

luc-github, 2025-03-06:
I use vscode with platformio under linux and setup is same as window for me

ahakkoca07, 2025-03-06:
I got problems like bottom blue bar not showing, crashing etc. before
now im downloading linux release

ahakkoca07, 2025-03-06:
in platformio.ini, do i need to edit this?
```
[env:include_tree]
platform         = atmelavr
board            = megaatmega2560
build_flags      = -c -H -std=gnu++11 -Wall -Os -D__MARLIN_FIRMWARE__
build_src_filter = +<src/MarlinCore.cpp>
```

ahakkoca07, 2025-03-06:
I installed platformio CLI, and like you said, it is so much easier to use than arduino IDE for compiling, it installed all the libraries automatically too! But i still have some warnings+i ran out of flash memory :(

[platformio.log](https://github.com/user-attachments/files/19111486/platformio.log)

I have a question, does it install the ESP3DLib from scratch?

ahakkoca07, 2025-03-06:
Ran it from VScode:
```
platformio run --silent -e esp32 ; echo "done" >|"/tmp/ipc"
ahakkoca07@mxlinux:~/Belgeler/GitHub/Marlin
$ platformio run --silent -e esp32 ; echo "done" >|"/tmp/ipc"
Marlin/src/inc/Warnings.cpp:67:4: warning: #warning "Your Configuration provides no method to acquire user feedback!" [-Wcpp]
   #warning "Your Configuration provides no method to acquire user feedback!"
    ^~~~~~~
Error: The program size (1402349 bytes) is greater than maximum allowed (1310720 bytes)
*** [checkprogsize] Explicit exit, status 1
================================================================================= [FAILED] Took 79.25 seconds =================================================================================

Environment    Status    Duration
-------------  --------  ------------
esp32          FAILED    00:01:19.245
============================================================================ 1 failed, 0 succeeded in 00:01:19.245 ============================================================================
ahakkoca07@mxlinux:~/Belgeler/GitHub/Marlin
$ 
```

ahakkoca07, 2025-03-06:
Fixed it by adding `board_build.partitions = Marlin/src/HAL/ESP32/esp32.csv` to ./Marlin/ini/esp32.ini, under `[env:esp32]`, now it compiles. But it still gives some warnings:
```

In file included from .pio/libdeps/esp32/WebSockets/src/WebSockets.cpp:45:
/home/ahakkoca07/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/mbedtls/port/include/esp32/sha.h:20:2: warning: #warning esp32/sha.h is deprecated, please use sha_parallel_engine.h instead [-Wcpp]
 #warning esp32/sha.h is deprecated, please use sha_parallel_engine.h instead
  ^~~~~~~
In file included from .pio/libdeps/esp32/ESP3DLib/src/modules/filesystem/esp_sd.cpp:37:
.pio/libdeps/esp32/SdFat/src/SdFat.h:452:2: warning: #warning File not defined because __has_include(FS.h) [-Wcpp]
 #warning File not defined because __has_include(FS.h)
  ^~~~~~~
In file included from .pio/libdeps/esp32/ESP3DLib/src/modules/filesystem/sd/sd_sdfat2_esp32.cpp:26:
.pio/libdeps/esp32/SdFat/src/SdFat.h:452:2: warning: #warning File not defined because __has_include(FS.h) [-Wcpp]
 #warning File not defined because __has_include(FS.h)
  ^~~~~~~
```

luc-github, 2025-03-06:
these are just warnings - they are not issues - I have already fixed them by using new version but this code it still in test

ahakkoca07, 2025-03-07:
Thank you, from now on i will be using VScode and PlatformIO for everything, it is so much easier as it maintains the libraries by itself. 
As my issue is resolved, i will be closing this issue, thank you 😄

closed 2025-03-07 by ahakkoca07

luc-github, 2025-03-07:
yes I provide a full setup which is compile , so it is easier for you ^_^