luc-github/ESP3DLib #88: [Question]Where platformio.ini file?

userosos, 2025-04-03:
Hi everyone! Where i can see platformio.ini file for compile it?

luc-github, 2025-04-03:
there is no platformio for esp3dlib, use Marlin, check my answer to your previous question

userosos, 2025-04-03:
> there is no platformio for esp3dlib, use Marlin, check my answer to your previous question

I check it but i can't understend how to i can install the ESP3DLib?
I can install marlin but after it? The libraru integrated in marlin?

luc-github, 2025-04-03:
Read the installation section,  yes esp3dlib is already integrated in Marlin

userosos, 2025-04-03:
> Read the installation section, yes esp3dlib is already integrated in Marlin

I see but it old version.

luc-github, 2025-04-03:
What is old version?

userosos, 2025-04-03:
> What is old version?

I want 3.0. In logs i can see **ESP3DLib @ 1.0.12**
May be because i use Marlin 2.1.2?

luc-github, 2025-04-03:
Read 3.0 installation you need my fork

userosos, 2025-04-03:
> Read 3.0 installation you need my fork

I foun the repo: https://github.com/luc-github/Marlin/tree/ESP3D-V3-2.1.x

But i can see the errors:

`Compiling .pio\build\mks_tinybee\libf96\ESP3DLib\core\espcmd\ESP107.cpp.o
In file included from .pio/libdeps/mks_tinybee/WebSockets/src/WebSockets.cpp:45:
C:/Users/user/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/mbedtls/port/include/esp32/sha.h:20:2: warning: #warning esp32/sha.h is deprecated, please use sha_parallel_engine.h instead [-Wcpp]
 #warning esp32/sha.h is deprecated, please use sha_parallel_engine.h instead
  ^~~~~~~
Compiling .pio\build\mks_tinybee\libf96\ESP3DLib\core\espcmd\ESP108.cpp.o
Compiling .pio\build\mks_tinybee\libf96\ESP3DLib\core\espcmd\ESP110.cpp.o
Archiving .pio\build\mks_tinybee\lib17a\libLittleFS.a
Compiling .pio\build\mks_tinybee\libf96\ESP3DLib\core\espcmd\ESP111.cpp.o
Archiving .pio\build\mks_tinybee\lib6d2\libSPIFFS.a
Compiling .pio\build\mks_tinybee\libf96\ESP3DLib\core\espcmd\ESP112.cpp.o
Compiling .pio\build\mks_tinybee\libf96\ESP3DLib\core\espcmd\ESP114.cpp.o
Archiving .pio\build\mks_tinybee\lib513\libWebSockets.a
In file included from .pio/libdeps/mks_tinybee/ESP3DLib/src/core/../include/../../../../../../Marlin/src/inc/MarlinConfig.h:28,
                 from .pio/libdeps/mks_tinybee/ESP3DLib/src/core/../include/../esp3dlib_config.h:34,
                 from .pio/libdeps/mks_tinybee/ESP3DLib/src/core/../include/esp3d_config.h:33,
                 from .pio/libdeps/mks_tinybee/ESP3DLib/src/core/benchmark.cpp:21:
.pio/libdeps/mks_tinybee/ESP3DLib/src/core/../include/../../../../../../Marlin/src/inc/MarlinConfigPre.h:52:51: fatal error: ../HAL/ESP32/inc/Conditionals_LCD.h: No such file or directory
   #include HAL_PATH(../HAL, inc/Conditionals_LCD.h)
                                                   ^
compilation terminated.
Compiling .pio\build\mks_tinybee\libf96\ESP3DLib\core\espcmd\ESP115.cpp.o
*** [.pio\build\mks_tinybee\libf96\ESP3DLib\core\benchmark.cpp.o] Error 1
In file included from .pio/libdeps/mks_tinybee/ESP3DLib/src/core/../include/../../../../../../Marlin/src/inc/MarlinConfig.h:28,
                 from .pio/libdeps/mks_tinybee/ESP3DLib/src/core/../include/../esp3dlib_config.h:34,
                 from .pio/libdeps/mks_tinybee/ESP3DLib/src/core/../include/esp3d_config.h:33,
                 from .pio/libdeps/mks_tinybee/ESP3DLib/src/core/commands.cpp:21:
.pio/libdeps/mks_tinybee/ESP3DLib/src/core/../include/../../../../../../Marlin/src/inc/MarlinConfigPre.h:52:51: fatal error: ../HAL/ESP32/inc/Conditionals_LCD.h: No such file or directory
   #include HAL_PATH(../HAL, inc/Conditionals_LCD.h)
                                                   ^
compilation terminated.
*** [.pio\build\mks_tinybee\libf96\ESP3DLib\core\commands.cpp.o] Error 1
In file included from .pio/libdeps/mks_tinybee/ESP3DLib/src/core/../include/../../../../../../Marlin/src/inc/MarlinConfig.h:28,
                 from .pio/libdeps/mks_tinybee/ESP3DLib/src/core/../include/../esp3dlib_config.h:34,
                 from .pio/libdeps/mks_tinybee/ESP3DLib/src/core/../include/esp3d_config.h:33,
                 from .pio/libdeps/mks_tinybee/ESP3DLib/src/core/debug_esp3d.cpp:21:
.pio/libdeps/mks_tinybee/ESP3DLib/src/core/../include/../../../../../../Marlin/src/inc/MarlinConfigPre.h:52:51: fatal error: ../HAL/ESP32/inc/Conditionals_LCD.h: No such file or directory
   #include HAL_PATH(../HAL, inc/Conditionals_LCD.h)
                                                   ^
compilation terminated.
*** [.pio\build\mks_tinybee\libf96\ESP3DLib\core\debug_esp3d.cpp.o] Error 1
In file included from .pio/libdeps/mks_tinybee/ESP3DLib/src/core/../include/../../../../../../Marlin/src/inc/MarlinConfig.h:28,
                 from .pio/libdeps/mks_tinybee/ESP3DLib/src/core/../include/../esp3dlib_config.h:34,
                 from .pio/libdeps/mks_tinybee/ESP3DLib/src/core/../include/esp3d_config.h:33,
                 from .pio/libdeps/mks_tinybee/ESP3DLib/src/core/esp3d.cpp:28:
.pio/libdeps/mks_tinybee/ESP3DLib/src/core/../include/../../../../../../Marlin/src/inc/MarlinConfigPre.h:52:51: fatal error: ../HAL/ESP32/inc/Conditionals_LCD.h: No such file or directory
   #include HAL_PATH(../HAL, inc/Conditionals_LCD.h)
                                                   ^
compilation terminated.
In file included from .pio/libdeps/mks_tinybee/ESP3DLib/src/core/../include/../../../../../../Marlin/src/inc/MarlinConfig.h:28,
                 from .pio/libdeps/mks_tinybee/ESP3DLib/src/core/../include/../esp3dlib_config.h:34,
                 from .pio/libdeps/mks_tinybee/ESP3DLib/src/core/../include/esp3d_config.h:33,
                 from .pio/libdeps/mks_tinybee/ESP3DLib/src/core/esp3doutput.cpp:21:
.pio/libdeps/mks_tinybee/ESP3DLib/src/core/../include/../../../../../../Marlin/src/inc/MarlinConfigPre.h:52:51: fatal error: ../HAL/ESP32/inc/Conditionals_LCD.h: No such file or directory
   #include HAL_PATH(../HAL, inc/Conditionals_LCD.h)
                                                   ^
compilation terminated.
In file included from .pio/libdeps/mks_tinybee/ESP3DLib/src/core/espcmd/../../include/../../../../../../Marlin/src/inc/MarlinConfig.h:28,
                 from .pio/libdeps/mks_tinybee/ESP3DLib/src/core/espcmd/../../include/../esp3dlib_config.h:34,
                 from .pio/libdeps/mks_tinybee/ESP3DLib/src/core/espcmd/../../include/esp3d_config.h:33,
[... cut ...]
========================================================================================= [FAILED] Took 82.29 seconds =========================================================================================

Environment    Status    Duration
-------------  --------  ------------
mks_tinybee    FAILED    00:01:22.293
==================================================================================== 1 failed, 0 succeeded in 00:01:22.293 ====================================================================================`

userosos, 2025-04-03:
I copy the folder into C:\ and i see the error:

```
Compiling .pio\build\mks_tinybee\lib84a\ESP3DLib\modules\http\handles\upload-updatefw.cpp.o
Compiling .pio\build\mks_tinybee\lib84a\ESP3DLib\modules\http\http_server.cpp.o
Compiling .pio\build\mks_tinybee\lib84a\ESP3DLib\modules\input\input.cpp.o
Compiling .pio\build\mks_tinybee\lib84a\ESP3DLib\modules\lua_interpreter\lua_interpreter_service.cpp.o
Compiling .pio\build\mks_tinybee\lib84a\ESP3DLib\modules\mDNS\mDNS.cpp.o
Compiling .pio\build\mks_tinybee\lib84a\ESP3DLib\modules\mks\mks_service.cpp.o
Compiling .pio\build\mks_tinybee\lib84a\ESP3DLib\modules\network\netconfig.cpp.o
Compiling .pio\build\mks_tinybee\lib84a\ESP3DLib\modules\network\netservices.cpp.o
.pio/libdeps/mks_tinybee/ESP3DLib/src/modules/http/handles/handle-description_xml.cpp:26:10: fatal error: ESP32SSDP.h: No such file or directory

*******************************************************************
* Looking for ESP32SSDP.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ESP32SSDP.h"
* Web  > https://registry.platformio.org/search?q=header:ESP32SSDP.h
*
*******************************************************************

 #include <ESP32SSDP.h>
          ^~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\mks_tinybee\lib84a\ESP3DLib\modules\http\handles\handle-description_xml.cpp.o] Error 1
.pio/libdeps/mks_tinybee/ESP3DLib/src/modules/network/netservices.cpp:37:10: fatal error: ESP32SSDP.h: No such file or directory

*******************************************************************
* Looking for ESP32SSDP.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ESP32SSDP.h"
* Web  > https://registry.platformio.org/search?q=header:ESP32SSDP.h
*
*******************************************************************

 #include <ESP32SSDP.h>
          ^~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\mks_tinybee\lib84a\ESP3DLib\modules\network\netservices.cpp.o] Error 1
================================================================================================= [FAILED] Took 104.17 seconds =================================================================================================

Environment    Status    Duration
-------------  --------  ------------
mks_tinybee    FAILED    00:01:44.170
============================================================================================ 1 failed, 0 succeeded in 00:01:44.170 ============================================================================================
```

userosos, 2025-04-03:
In platformio.ini in [common] section i added:

 lib_deps           = ESP32SSDP
And it compiled now.

luc-github, 2025-04-03:
I have too many version live and missed to update the ini file due to new platformio syntax
Update the line in features.ini :

ESP32SSDP=luc-github/ESP32SSDP@1.2.0

by 

ESP32SSDP=https://github.com/luc-github/ESP32SSDP/archive/refs/tags/1.2.2.zip

I will clean up all versions soon - it become too messy now