luc-github/ESP3DLib #89: [Question]Do Telnet have answer?

userosos, 2025-04-04:
Does Telnet be able to return the response from G-Code?
I send the teams to the ESP3DLIB Teln, they reach perfectly, but do not return anything. Can Telnet be taught to return the commands? I did not find WebSocket in ESP3DLIB V3.0
I send the M114 command to Teln and expect an answer: X: 0.00 Y: 0.00 Z: 170.00 E: 0.00 Count X: 0 Y: 0 Z: 15111016 in Telnet (I Can See in Terminal Esp3web)

userosos, 2025-04-04:
I use node-red with telnet node:
https://flows.nodered.org/node/node-red-contrib-telnet-client
Node telnet-request.

In debug Node-red i can see only:
```
[node: esp3dlib telnet](http://192.168.55.55:8123/api/hassio_ingress/BrGHxjrjy_dIbBzGpmGozYZoXYwc-PEZX16d2ALfbzc/#)
msg : string[51]
"Response wait exceeded timeout value (request 5000)"
```

> Yes commands send by telnet return answer
> what client do you use to connect to 192.168.55.156:23 ?

userosos, 2025-04-07:
> 192.168.55.156:23 is not real telnet is more raw tcp, telnet is confusing name used by esp sample code sorry try using Putty using Raw connection, or your application may have this feature

Good! I use tcp requests node - work fine with answer from esp3dlib. Big THX! The software has big help In scientific work! Great software.

luc-github, 2025-04-07:
Ho really happy to learn that thank you, what are you doing if I may ask ?

userosos, 2025-04-07:
> Ho really happy to learn that thank you, what are you doing if I may ask ?

We use this to automatically change the position of our vacuum chamber samples. In general, we are engaged in spectroscopy,  explore new materials.

luc-github, 2025-04-07:
Nice I used to work with IR spectroscopy in Ink control lab then in Veterinary analysis Labs, was long time ago haha